hardbyte/python-can #379: ASC log header has duplicate lines

bauer-san, 2018-07-31:
The `base hex  timestamps absolute` line is written twice by the ASC log writer.  Line 136 in __init__ and 166 are each writing these lines.  The resultant .ASC file cannot be read by Vector tools.

```
date Tue Jul 07 09:49:50 AM 2018
base hex  timestamps absolute
internal events logged
base hex  timestamps absolute
Begin Triggerblock Tue Jul 07 09:49:50 AM 2018
```

felixdivo, 2018-07-31:
Yes, I see the problem in the code. I could if fix it as part of the huge PR #348, but I need to know whether the first or the last line with `base hex ...` is wrong.

felixdivo, 2018-07-31:
In the example in `python-can/test/data/logfile.asc`, the second line is missing.

bauer-san, 2018-07-31:
Exactly right.  
I should have stated the expected behavior in the original issue.

felixdivo, 2018-07-31:
No problem. I'll fix it right now, and the patch will be available on the next release after #348 got merged.

closed 2018-08-06 by felixdivo