hardbyte/python-can #386: Error during setup wrapt

ivomans, 2018-08-07:
On a new Raspbian image setup.py aborts with underneath message. Unfortunately I'm not an expert on pypi usage. Hope someone can help out.
```
Searching for wrapt~=1.10
Reading https://pypi.python.org/simple/wrapt/
Download error on https://pypi.python.org/simple/wrapt/: unknown error (_ssl.c:3172) -- Some packages may not be found!
Couldn't find index page for 'wrapt' (maybe misspelled?)
Scanning index of all packages (this may take a while)
Reading https://pypi.python.org/simple/
Download error on https://pypi.python.org/simple/: unknown error (_ssl.c:3172) -- Some packages may not be found!
No local packages or working download links found for wrapt~=1.10
error: Could not find suitable distribution for Requirement.parse('wrapt~=1.10')

```

felixdivo, 2018-08-07:
Hm, strange. https://pypi.python.org/simple/wrapt/ contains the version 1.10. How did you invoke the script or how did you install python-can?

felixdivo, 2018-08-07:
Hm, maybe `sudo pip3 install -vvv wrapt` will give some more useful error message?

ivomans, 2018-08-08:
The -vvv option learned me there was a problem with curl.
Turned out any command on commandline with curl opening a https url gave an error:
```
curl: (77) error setting certificate verify locations:
  CAfile: /etc/ssl/certs/ca-certificates.crt
  CApath: /etc/ssl/certs
```
Found somewhere a simple solution to that: 
`apt-get install --reinstall ca-certificates`
Apparently a problem in the current Raspbian image.
Your setup.py now finishes without a problem. Thanks for your help.

felixdivo, 2018-08-08:
Okay, great! 😄

closed 2018-08-08 by felixdivo