hardbyte/python-can #391: Simple Notifier/Listener example?

mindfuucker, 2018-08-15:
Hello,

I'm trying to use the python-can but am having difficulties creating a listener. 

Could you tell me what i'm doing wrong?

`def parse_data(single_can_frame):
    print(single_can_frame)

bus = can.interface.Bus("can0", bustype="socketcan")
notifier = can.Notifier(bus, [parse_data(can.message)])

while True:
    pass`

closed 2018-08-17 by felixdivo

mindfuucker, 2018-08-23:
Thank you @felixdivo 

I implemented my own listener to solve it.

```
class Parse_Data_Listener(can.Listener):
    def on_message_received(self, my_single_message):
```

felixdivo, 2018-08-23:
Okay, nice to hear that the problem is solved. If you have suggestions on how to improve the docs, please feel free to send a PR or open a issue with a concrete problem.