hardbyte/python-can #392: Problem with new location of scripts

felixdivo, 2018-08-15:
> Calling: `python -m can.logger` does not seem to work on the develop branch? I have to use the full path: `python -m can.scripts.logger`. 

Reported by @Lauszus [over here](https://github.com/hardbyte/python-can/pull/370#issuecomment-412303400).

felixdivo, 2018-09-01:
Also need to consider newly added script from #390.

felixdivo, 2018-09-04:
(Note for myself: TODO: Update `setup.cfg`.)

felixdivo, 2018-09-04:
I added some regression tests to branch `python-can/fix_scripts`, and will merge it together with the fix of this issue once we decided what to do.

christiansandberg, 2018-09-05:
I would prioritize user interface and external API over internal structure, but that's just my opinion. I didn't see the way it was before as a big issue.

felixdivo, 2018-09-05:
Well, I guess you're right. Im my opinion it was confusing for people new to the codebase, but that's just how it is now. I'll revert the relevant changes. -.-

PR #406 (2018-09-06): Fix location of scripts
1. Placing the scripts in a separate module in `can.modules.*` instead of `can.*` made them impossible to be called the old way, even with them being present in the `can/__init__.py`. Thus, the relocation of the scripts to the separate module is reverted.
1. Now, the scripts print the help message when no arguments were given.
1. Adds some tests to ensure that the scripts can always be called.

Fix #392
closed 2018-09-06 by felixdivo