hardbyte/python-can #395: socketcan issue

sathyadurai, 2018-08-23:
can_bus = can.interface.Bus(bustype='socketcan', channel=can_recv_port) 
can_bus.send()

I'm using this combination to run in a while loop for my testing. This creates too many sockets in the system that halts it after about an hour. 

can_bus.shutdown() is not helping either.

yegorich, 2018-08-23:
What about maintaining `can_bus`  throughout the tests?

felixdivo, 2018-08-23:
Well, what exactly does the while loop look like? It the above code inside the loop? You need to either use the same bus for all iterations, or et least close the bus. You can use it like a context manager with should be simpler and more robust. Example:

```python
can_recv_port = "vcan0"
with can.interface.Bus(bustype='socketcan', channel=can_recv_port) as can_bus:
    while (some_stuff == True):
        can_bus.send(...)
```

Does that solve you problem?

closed 2018-08-30 by felixdivo