hardbyte/python-can #396: How to exit the python script?

mindfuucker, 2018-08-23:
Hello everyone,

I have the following script:

```
import can
import requests
import sys
import struct

class Parse_Data_Listener(can.Listener):
    def on_message_received(self, my_single_message):
        timestamp = my_single_message.timestamp
        unpacked_data = struct.unpack('>bhhhb', my_single_message.data)
        payload = {'timestamp': timestamp, 'voltage': unpacked_data[1]}
        url = WEBSERVER_IP + '/addVoltage'
        try:
            r = requests.post(url, data=payload)
        except requests.exceptions.RequestException as e:
            print(e)
            sys.exit(2)

print("Connecting to server")
try:
    response = requests.get(WEBSERVER_IP)
except:
    print("No connection to: " + WEBSERVER_IP)
    sys.exit(3)
    pass
else:
    my_listener = Parse_Data_Listener()
    bus = can.interface.Bus("can0", bustype="socketcan")
    notifier = can.Notifier(bus, [my_listener])
```

When connecting to the server initially fails, it nicely exits with error-code 3.
When already connected and the connection fails it doesn't exit nicely. It prints the error but stays active. Is this a bug of the notifier or just a PEBKAC?

hardbyte, 2018-09-01:
My hunch is the main thread is terminating while a daemon thread is still doing work. In some of our examples and scripts we have a `while True: sleep` loop just to stay alive waiting for the Ctrl-C signal - https://github.com/hardbyte/python-can/blob/develop/examples/receive_all.py#L17

mindfuucker, 2018-09-01:
I am indeed using a 

```
while True:
    pass
```

To keep te listener active. Would sleep be better? 
I eventually fixed it by calling os._exit().

I'm running my script with supervisor so I needed to be certain it would exit on error.

felixdivo, 2018-09-06:
`sleep()` would be better. It does not constantly require the CPU to work, but instead actually sleeps. But please sleep for more than a few milliseconds, see [here](https://stackoverflow.com/a/10926766/3753684).

felixdivo, 2018-09-06:
And please call `bus.shutdown()` in a try-finally block or use a context manager syntax to free any resources.

closed 2018-09-06 by felixdivo

mindfuucker, 2018-09-06:
@felixdivo, how do I reference the bus inside its listener?

self.getbus()

felixdivo, 2018-09-06:
Well, the listener does not hold a reference to the bus. Instead, it gets called by the bus each time there is a message. You can store the bus variable somewhere globally, or even better make it a member of the listener:

```Python
with Bus(interface="...", ...) as bus:
    listener = Parse_Data_Listener() # or add it in the constructor
    listener.bus = bus
    # add it to a notifier
```

mindfuucker, 2018-09-06:
@felixdivo Thank you for your expertise

```
class Parse_Data_Listener(can.Listener):
    def __init__(self, bus):
        self.bus = bus

    def on_message_received(self, my_single_message):
         try:
            r = requests.post(url, data=payload)
        except requests.exceptions.RequestException as e:
            print(e)
            self.bus.shutdown()
            sys.exit(2)
```

felixdivo, 2018-09-06:
Looks good!