hardbyte/python-can #398: Specifying handshake for serial port based CAN interfaces

yegorich, 2018-08-28:
So far we can only pass baudrate to the interface constructor like `/dev/ttyUSB0@115200`. Many adapters run at high baudrates so enabling hardware handshake is a must. There are two issues:

1. what handshake options do we need to support: RTS/CTS, Xon/Xoff etc.
2, depending on the answer to the first issue the syntax must be defined. For example: `/dev/ttyUSB0@115200:rtscts` or something like this.

felixdivo, 2018-08-30:
Slightly off-topic: Currently #290 is blocking some of the effort on the serial interface I think.

yegorich, 2018-08-30:
AFAIK #290 will have to be reworked anyway. And introducing handshake support won't interfere with it. I.e. the changes will just have to be ported. At least we can define how and what should be implemented regarding the handshake.

felixdivo, 2018-08-30:
Yup, that right. I am not familiar with the serial interface though.

felixdivo, 2018-08-30:
Ah, I see you already commented on the topic of this issue [over there](https://github.com/hardbyte/python-can/pull/290#issuecomment-379528855).

yegorich, 2018-08-30:
Yes, and as #290 got stuck I've decided at least to implement the features needed for our products.

I'll prepare a PR implementing `/dev/ttyUSB0@115200:rtscts`.

PR #402 (2018-09-04): Add hardware handshake support for serial port based CAN interfaces
pySerial module provides 'rtscts' option to handle hardware handshake.
This setting will be extracted from __init__'s kwargs.

Example:

```Python
bus = can.interface.Bus(bustype='slcan',
                                      channel='/dev/ttyUSB0@3000000',
                                      bitrate=1000000,
                                      rtscts=True)
```

Closes #398
closed 2018-09-04 by felixdivo