hardbyte/python-can #401: Long shot: async listener?

mindfuucker, 2018-08-31:
Hello everyone,

First I like to say that In really like the library and have been using it extensively the past month. Especially the notifier/listener construction works for me. 

I'm trying to do some work (parsing & HTTP-request) inside the listener, this creates a bottleneck that I need to fix. CAN-messages are now being lost. To fix it I am going to implement an asynchronous method/function. 

I am not sure of this is the bottleneck so I have some questions:

- What happens if one listener is still active when another message arrives?
- Has anyone done this before with this library?
- Could I request this as a feature for upcoming releases?

christiansandberg, 2018-08-31:
You shouldn't do any blocking operations inside the listener as that will stop reception of messages. So either you have to off-load the blocking operation to a separate thread, or you can wait for my #388 pull request which I made to solve exactly these types of tasks without having to manage threads.

mindfuucker, 2018-08-31:
@christiansandberg 
Wow, that look impressive. I did not notice someone already implementing this. 
Thank you for your expertise about blocking operations.

felixdivo, 2018-08-31:
> You shouldn't do any blocking operations inside the listener as that will stop reception of messages.

Maybe we should document that somewhere, or has that already been done?

christiansandberg, 2018-09-01:
However, I don’t think this will solve your problem. Making individual HTTP requests is slow, so in order to not just building up a growing backlog of requests you probably need to switch to something better suited for real time communication like WebSockets.

mindfuucker, 2018-09-01:
@christiansandberg 

Thank you, I will have a look into websockets. Would MQQT also be an option?

christiansandberg, 2018-09-01:
MQTT would also work. Basically anything that has a persistent connection and not too much overhead.