hardbyte/python-can #404: CyclicSendTask can no longer restart with socketcan

hardbyte, 2018-09-01:
The `CyclicSendTask` which inherits from `RestartableCyclicTaskABC` closes the bcm socket when the tasks is stopped. 

A replacement socket is created when the task is restarted... but if the user wants to modify the data while the task is stopped the `modify_data` method tries to use the closed bcm socket raising an exception (e.g. run `examples/cyclic.py`:

```
can.CanError: Couldn't send CAN BCM frame. OS Error 9: Bad file descriptor
The CAN socket appears to be closed.
```

I admit this is quite a corner case but perhaps we should keep the socket open until the task is garbage collected? What do you think @christiansandberg ?

christiansandberg, 2018-09-01:
Oh, that's right. What do you think about having one BCM socket per bus instead? That would solve #292 too.

christiansandberg, 2018-09-06:
It's only the socketcan interface that suffers from the garbage collector problem. Threads keep running even if there is no reference to it (but I'm not sure if that is a documented behaviour and they will fail ungracefully when the bus is shutdown). The IXXAT interface just tells the hardware to send a message periodically.

felixdivo, 2018-09-06:
Okay, if you say so!

felixdivo, 2018-09-28:
> Both #409 and #412 seem fine to close this.

Closing because #412 got merged.

closed 2018-09-28 by felixdivo