hardbyte/python-can #416: esdcan

whyAtGh, 2018-09-14:
Could python-can support the esdcan interface? And what's the interface name?

felixdivo, 2019-02-11:
The interface name is generally just a well memorable short string. Generally, extensions could be written for almost any type of interface. You are very welcome to contribute it, and we will help you get started if you want to. Until then, I will close this issue.

DanseMacabre, 2019-03-04:
I've had need to write a driver for ESD's CAN cards utilising their provided python bindings.

It's not in any way ready for PR, but feedback on it would be appreciated, especially regarding the limitation ESD seems to add to their timeouts?

See this gist for the driver as written: https://gist.github.com/DanseMacabre/60f26aa4abed4d4b3bb8a9aacf439fd1

felixdivo, 2019-03-04:
I am currently a bit short of time. But you could open a PR and I could tag it as "work in progress". It will probably get some attention from others as well so we can work on this together.

But I just wanted to link these two sources for you
https://python-can.readthedocs.io/en/stable/development.html#creating-a-new-interface-backend
https://python-can.readthedocs.io/en/stable/internal-api.html#businternals
But it seems like you have already followed it. 😄

As for the name of the interface, both `esdcan` and `ntcan` seem OK, I guess this is subjective.

At the end, you can also leave a note in `python-can/doc/history.rst`.

Some first feedback:
- `self.RxQS = 2000` (and alike, as well as `self.CIF`) should be lowercase with underscores, see [PEP8](https://www.python.org/dev/peps/pep-0008/#function-and-variable-names).
- `_recv_internal` should only return `True` as a second parameter if filtering actually took place. Else return `False` to let the superclass fall back to software filtering.
- Could `_recv_internal` also pass `is_extended_id` as a parameter to the message constructor?

I did not have a look at the docs from esd GmbH, are they available somewhere else more freely?

DanseMacabre, 2019-03-05:
I do believe the only place to get the docs is ESD GmbH. The docs are "free" to download, and only requires an account on their website to access. The latest revision available is revision 5.2 on ESD GmbH. A revision 4.3 is available from their US (branch?) here: http://www.esd-electronics-usa.com/Shared/Docu/CANPCI400/CAN-API_Part1_Function_Manual_43.pdf . The differences seem to be largely CAN-FD related things, along side a couple of new features such as timestamped messages etc.

gohierf, 2024-02-01:
For the record, PyNTCAN is now shared as wheel files for python 3.10 and 3.11, in their CAN SDK (`C:\Program Files\ESD\CAN\SDK\lib\Python`) installation.