hardbyte/python-can #418: Packaging for Fedora Linux

cottsay, 2018-09-20:
This package is a candidate for inclusion in the Fedora Linux repositories. There are a couple of things that should happen to make this clean. We can work around the issues in packaging if you don't want to take the changes, but they're probably goodness outside of Fedora anyway.

- [ ] Wrong EOL in several files (I will open a PR)
- [ ] Shebang used in many non-executable scripts, and vice-versa
- The example scripts are inconsistent, but the resolution is clear (I will open a PR)
- Pretty much everything in the module has shebangs even though the files aren't intended to be executed. Unless there is a reason, they should be removed. Once the module is installed, the any executable permissions appear to be dropped anyway, rendering them useless. Note that the presence of a shebang has no effect on the ability to invoke the script directly from a python executable (e.g. `python3 foo.py`)

christiansandberg, 2018-09-20:
I agree on the shebang lines. Just for background it was introduced in #259 and discussed briefly [here](https://github.com/hardbyte/python-can/pull/255#discussion_r174899569).

felixdivo, 2018-09-20:
I guess pip is installed on Fedora, and `python-can` can be used by downloading from PyPi anyways. What is the benefit of providing it as an extra package?

cottsay, 2018-09-20:
The same benefits as there would be for DEBs or AUR or any other distribution's package management system apply to RPMs in Fedora. The motivation for me to onboard this package in particular is so that other Fedora packages can take a dependency on it.

Availability as an RPM doesn't preclude installation via pip, but offers a curated alternative that was packaged to Fedora packaging specs (hence this issue). It also makes the package available for pre-built "spins" of Fedora, which don't include pip packages.

Thanks for jumping on this quickly - its so much more pleasant to work with responsive maintainers! I'm glad to hear that the shebang issue has come up before - my opinion is that the filenames make it clear that the files are python, and that they don't really contain any version information. This is especially true given that your package is functional in both python 2 and 3. If you'd like, I can open a PR to remove them, but I wanted to discuss it first.

felixdivo, 2018-09-20:
Well, sounds reasonable! (Although including PyPI packages as rpm dependencies would be way more awesome and simpler for maintainers, but yeah - I know - in practice that's tricky. 😛)

About the Shebangs; I guess there isn't a "right or wrong", it rather doesn't really matter that much. If the Fedora package specs requires the removal, I'm okay with that. 

How do we ensure that the Fedora package stays up to date with the PyPI package?

cottsay, 2018-09-20:
Fedora leverages Anitya [[1](https://release-monitoring.org/)] to ensure upstream updates get noticed by maintainers in the Fedora build system. In practice, this is WAY easier for packages which use git tags and have tests that we can run automatically. The less stuff we need to work around on our end, the smoother package updates become.

They way this will look is that Anitya will notice the release and a ticket will be filed in RedHat's Bugzilla assigned to the point-of-contact for the package (me). The system will probably automatically try to build and test the new package to get a feel for how easy the upgrade will be. Then the assignee will make the call as to whether the package is updated only for Fedora v-next, or if the update is safe to bring back to stable releases. Typically, the only time an update isn't made available to stable releases is if there was a major version bump or other breaking changes to the API that might jeopardize the stability of any downstream consumers.

re: shebangs - Fedora doesn't so much *require* that the linter errors be fixed upstream, but they typically come up for a good reason and provide benefit to the larger community. My responsibility is just to make you aware of the issues :slightly_smiling_face: If you're interested, There's a lot more info on packaging and shebang handling specifically on Fedora's wiki [[2](https://fedoraproject.org/wiki/Packaging:Guidelines#Shebang_lines)].

Thanks again for your quick response!

[1] https://release-monitoring.org/
[2] https://fedoraproject.org/wiki/Packaging:Guidelines#Shebang_lines

felixdivo, 2018-09-23:
Sounds good!

cottsay, 2018-09-27:
Thanks for working with me on these fixes! I'll start moving forward with the review process. When that is done, I'll ping back here so we can close this issue.

hardbyte, 2018-12-24:
Any updates @cottsay?

closed 2018-12-24 by hardbyte