hardbyte/python-can #421: Where is the Notifier timeout raised?

mindfuucker, 2018-09-21:
Hello everybody,
I'm trying to implement a listener that shuts down the whole script as CAN-messages are not send for a minute. 
What I want to know is where the timout is raised?: 

- Inside \_\_main__
- the while-loop 
- the listener 
- the thread created by the listener 

**This is my code:**
```
class Parse_Data_Listener(can.Listener):
    def __init__(self):
        self.websocket = None

    def on_message_received(self, my_single_message):
        t1 = threading.Thread(target=self.this_runs_in_thread, args=(my_single_message,))
        t1.start()

    def this_runs_in_thread(self, my_single_message):
        ws = self.websocket
        canid = my_single_message.arbitration_id
        if canid in BATTERY:
            func = BATTERY[canid]
            func.parse_and_send_message(ws, my_single_message)
        else:
            #print(f'not found: {canid} - {my_single_message}')

if __name__ == "__main__":
    print("Connecting to server")
    try:
        response = requests.get("http://"+WEBSERVER_IP, verify=False, timeout=5)
    except:
        print("No connection to: " + WEBSERVER_IP)
        sys.exit(3)
    else:
        hostname = socket.gethostname()
        my_listener = Parse_Data_Listener()
        my_listener.websocket = websocket.create_connection("ws://"+WEBSERVER_IP+"/ws/chat/"+hostname+"/")
        bus = can.interface.Bus("can0", bustype="socketcan")
        notifier = can.Notifier(bus, [my_listener], timeout=60)

        while True:
            pass
```

hardbyte, 2018-09-22:
The `timeout` that is passed to a `Notifier` doesn't raise an exception if no message is received in that time. Rather the timeout is passed into the bus.recv function and could be useful in some backends to avoid blocking. Since you're doing things with bare threads I'd consider creating a `Timer` which you reset every time you get a message.

mindfuucker, 2018-09-22:
@hardbyte 
Thank you, then I misread the docs. I will create timer that checks periodically.

closed 2018-09-22 by mindfuucker