hardbyte/python-can #431: Install failure

tthatche, 2018-09-26:
I'm having issues using PyPi (pip) to install python-can with both python 2.7.9 and with 3.4.2.   It appears that the issue in both cases is the "install_requires" line.  I don't have enough experience in python to be able to pull request and fix though.  

Debian GNU/Linux 8 ts-imx6ul ttymxc0

ts-imx6ul login: root
Last login: Tue Sep 25 18:03:32 UTC 2018 on ttymxc0
Linux ts-imx6ul 4.1.15-g7aa498a #16 SMP PREEMPT Fri Dec 15 13:19:46 PST 2017 arm                      v7l

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.

root@ts-imx6ul:/usr/lib/python3.4# pip3 list
atomicwrites (1.2.1)
attrs (18.2.0)
chardet (2.3.0)
colorama (0.3.2)
html5lib (0.999)
mock (2.0.0)
more-itertools (4.3.0)
nose (1.3.7)
pathlib2 (2.3.2)
pbr (4.2.0)
pip (1.5.6)
pluggy (0.7.1)
py (1.6.0)
pyserial (3.4)
pytest (3.8.1)
pytest-timeout (1.3.2)
requests (2.4.3)
scandir (1.9.0)
setuptools (5.5.1)
six (1.11.0)
urllib3 (1.9.1)
wheel (0.24.0)
root@ts-imx6ul:/usr/lib/python3.4# uname -r
4.1.15-g7aa498a
root@ts-imx6ul:/usr/lib/python3.4# pip3 install python-can
Downloading/unpacking python-can
  Downloading python_can-2.2.1-py2.py3-none-any.whl (136kB): 136kB downloaded
Cleaning up...
Exception:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2595, in _dep_map
    return self.__dep_map
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2457, in __getattr__
    raise AttributeError(attr)
AttributeError: _DistInfoDistribution__dep_map

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/commands/install.py", line 290, in run
    requirement_set.prepare_files(finder, force_root_egg_info=self.bundle, bundle=self.bundle)
  File "/usr/lib/python3/dist-packages/pip/req.py", line 1266, in prepare_files
    req_to_install.extras):
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2401, in requires
    dm = self._dep_map
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2597, in _dep_map
    self.__dep_map = self._compute_dependencies()
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2621, in _compute_dependencies
    parsed = next(parse_requirements(distvers))
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2721, in parse_requirements
    "version spec")
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2686, in scan_list
    raise ValueError(msg, line, "at", line[p:])
ValueError: ('Expected version spec in', 'wrapt ~=1.10', 'at', ' ~=1.10')

Storing debug log for failure in /root/.pip/pip.log
root@ts-imx6ul:/usr/lib/python3.4# python3 --version
Python 3.4.2
root@ts-imx6ul:/usr/lib/python3.4#

hardbyte, 2018-09-27:
Thanks for the bug report. As it happens we've been preparing a new release which may fix this already.

Could you please try:

```
pip install python-can==2.3.0a1
```

tthatche, 2018-09-27:
Downloading/unpacking python-can==2.3.0a1
  Downloading python-can-2.3.0a1.tar.gz (128kB): 128kB downloaded
  Running setup.py (path:/tmp/pip-build-zp1y4cjp/python-can/setup.py) egg_info for package python-can
    /usr/lib/python3.4/distutils/dist.py:260: UserWarning: Unknown distribution option: 'python_requires'
      warnings.warn(msg)
    error in python-can setup command: 'tests_require' must be a string or list of strings containing valid project/version requirement specifiers
    Complete output from command python setup.py egg_info:
    /usr/lib/python3.4/distutils/dist.py:260: UserWarning: Unknown distribution option: 'python_requires'

  warnings.warn(msg)

error in python-can setup command: 'tests_require' must be a string or list of strings containing valid project/version requirement specifiers

----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in /tmp/pip-build-zp1y4cjp/python-can
Storing debug log for failure in /root/.pip/pip.log
root@ts-imx6ul:/usr/lib/python3.4/site-packages/SRPyLogger#

hardbyte, 2018-09-28:
I've pushed a version without the spaces if you wouldn't mind trying again with:

```
pip install python-can==2.3.0a2
```

tthatche, 2018-09-28:
Crazy-- Still failing. 

<ckages/SRPyLogger# pip3 install python-can==2.3.0a1
Downloading/unpacking python-can==2.3.0a1
  Downloading python-can-2.3.0a1.tar.gz (128kB): 128kB downloaded
  Running setup.py (path:/tmp/pip-build-p5sm3r9n/python-can/setup.py) egg_info for package python-can
    /usr/lib/python3.4/distutils/dist.py:260: UserWarning: Unknown distribution option: 'python_requires'
      warnings.warn(msg)
    error in python-can setup command: 'install_requires' must be a string or list of strings containing valid project/version requirement specifiers
    Complete output from command python setup.py egg_info:
    /usr/lib/python3.4/distutils/dist.py:260: UserWarning: Unknown distribution option: 'python_requires'

  warnings.warn(msg)

error in python-can setup command: 'install_requires' must be a string or list of strings containing valid project/version requirement specifiers

----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in /tmp/pip-build-p5sm3r9n/python-can
Storing debug log for failure in /root/.pip/pip.log
root@ts-imx6ul:/usr/lib/python3.4/site-packages/SRPyLogger#

tthatche, 2018-09-28:
oops - i tried installing the old version.  Here is the result from the new build.

<ckages/SRPyLogger# pip3 install python-can==2.3.0a2
Downloading/unpacking python-can==2.3.0a2
  Downloading python_can-2.3.0a2-py2.py3-none-any.whl (122kB): 122kB downloaded
Cleaning up...
Exception:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2595, in _dep_map
    return self.__dep_map
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2457, in __getattr__
    raise AttributeError(attr)
AttributeError: _DistInfoDistribution__dep_map

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/commands/install.py", line 290, in run
    requirement_set.prepare_files(finder, force_root_egg_info=self.bundle, bundle=self.bundle)
  File "/usr/lib/python3/dist-packages/pip/req.py", line 1266, in prepare_files
    req_to_install.extras):
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2401, in requires
    dm = self._dep_map
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2597, in _dep_map
    self.__dep_map = self._compute_dependencies()
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2621, in _compute_dependencies
    parsed = next(parse_requirements(distvers))
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2721, in parse_requirements
    "version spec")
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2686, in scan_list
    raise ValueError(msg, line, "at", line[p:])
ValueError: ('Expected version spec in', 'wrapt ~=1.10', 'at', ' ~=1.10')

Storing debug log for failure in /root/.pip/pip.log
root@ts-imx6ul:/usr/lib/python3.4/site-packages/SRPyLogger#

hardbyte, 2018-09-28:
Ok this is installing for me on a raspberry pi now.

```
pip3 install -U python-can==3.0.0-a1
```

tthatche, 2018-09-28:
<hon3.4/site-packages/python-can# pip3 install -U python-can==3.0.0-a1
Downloading/unpacking python-can==3.0.0-a1
  Cannot fetch index base URL https://pypi.python.org/simple/
  Could not find any downloads that satisfy the requirement python-can==3.0.0-a1
Cleaning up...
No distributions at all found for python-can==3.0.0-a1
Storing debug log for failure in /root/.pip/pip.log
root@ts-imx6ul:/usr/lib/python3.4/site-packages/python-can#

tthatche, 2018-09-28:
my hardware is an embedded arm board.  TS7553 V2.  Should be very similiar to RPi

https://www.embeddedarm.com/products/TS-7553-V2

hardbyte, 2018-09-28:
The installation command should be `pip install -U python-can==3.0.0a1`

jmpentney, 2018-09-29:
Same problem here. Trying to install on a Techbase NPE X500, which is an industrialized Raspberry Pi running Debian Jessie (though with a custom kernel):

http://www.techbase.eu/en/products/npe-x500.html

The output of pip install is:

root@techbase:~# pip install -U python-can==3.0.0a1
Downloading/unpacking python-can==3.0.0a1
  Downloading python_can-3.0.0a1-py2.py3-none-any.whl (122kB): 122kB downloaded
Cleaning up...
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 290, in run
    requirement_set.prepare_files(finder, force_root_egg_info=self.bundle, bundle=self.bundle)
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 1266, in prepare_files
    req_to_install.extras):
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2401, in requires
    dm = self._dep_map
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2597, in _dep_map
    self.__dep_map = self._compute_dependencies()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2621, in _compute_dependencies
    parsed = next(parse_requirements(distvers))
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 49, in <lambda>
    next = lambda o: o.next()
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2721, in parse_requirements
    "version spec")
  File "/usr/lib/python2.7/dist-packages/pkg_resources.py", line 2686, in scan_list
    raise ValueError(msg, line, "at", line[p:])
ValueError: ('Expected version spec in', 'wrapt ~=1.10', 'at', ' ~=1.10')

Storing debug log for failure in /root/.pip/pip.log

I've tried installing wrapt independently but it makes no difference to the installation of python-can. I've also tried installing Python 3.4 and pip3, but get the same problem.

Any help would be gratefully appreciated!

hardbyte, 2018-09-30:
Appreciate the help to make sure we don't release until this is working. 

From your traceback it really looks like you're still running into the space we had in the version specification. I triple checked the requirements and pushed a new version. I tested installing from PyPi on Python3.5 and Python2.7 (using pipenv and virtualenv on a Raspberry Pi) and it is working for me.

Could you please try this again `pip install -U python-can==3.0.0-a2`

For reference this is my output with pipenv:
```
pi@node-3:~ $ mkdir test-can-3.0a2-py35
pi@node-3:~ $ cd test-can-3.0a2-py35/
pi@node-3:~/test-can-3.0a2-py35 $ python3 -m pipenv --python 3.5
Creating a virtualenv for this project...
Pipfile: /home/pi/test-can-3.0a2-py35/Pipfile
Using /usr/bin/python3.5m (3.5.3) to create virtualenv...
⠋Running virtualenv with interpreter /usr/bin/python3.5m
Using base prefix '/usr'
New python executable in /home/pi/.local/share/virtualenvs/test-can-3.0a2-py35-LXcdVefF/bin/python3.5m
Also creating executable in /home/pi/.local/share/virtualenvs/test-can-3.0a2-py35-LXcdVefF/bin/python
Installing setuptools, pip, wheel...done.
Setting project for test-can-3.0a2-py35-LXcdVefF to /home/pi/test-can-3.0a2-py35

Virtualenv location: /home/pi/.local/share/virtualenvs/test-can-3.0a2-py35-LXcdVefF
Creating a Pipfile for this project...                                                                                                          pi@node-3:~/test-can-3.0a2-py35 $ python3 -m pipenv shell
Launching subshell in virtual environment…
 . /home/pi/.local/share/virtualenvs/test-can-3.0a2-py35-LXcdVefF/bin/activate
pi@node-3:~/test-can-3.0a2-py35 $  . /home/pi/.local/share/virtualenvs/test-can-3.0a2-py35-LXcdVefF/bin/activate
(test-can-3.0a2-py35-LXcdVefF) pi@node-3:~/test-can-3.0a2-py35 $ pip install -U python-can==3.0.0-a2
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting python-can==3.0.0-a2
  Using cached https://files.pythonhosted.org/packages/ff/b3/d4ea95bb2d7367cc0d86fa11110d2bf10368e2947ad57f72e5583b24bd85/python_can-3.0.0a2-py2.py3-none-any.whl
Collecting wrapt~=1.10 (from python-can==3.0.0-a2)
  Using cached https://www.piwheels.org/simple/wrapt/wrapt-1.10.11-cp35-cp35m-linux_armv7l.whl
Collecting typing (from python-can==3.0.0-a2)
  Using cached https://files.pythonhosted.org/packages/4a/bd/eee1157fc2d8514970b345d69cb9975dcd1e42cd7e61146ed841f6e68309/typing-3.6.6-py3-none-any.whl
Installing collected packages: wrapt, typing, python-can
Successfully installed python-can-3.0.0a2 typing-3.6.6 wrapt-1.10.11
```

tthatche, 2018-10-01:
<ges/python-can# pip3 install -U python-can==3.0.0-a2
Downloading/unpacking python-can==3.0.0-a2
  Downloading python_can-3.0.0a2-py2.py3-none-any.whl (123kB): 123kB downloaded
Cleaning up...
Exception:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2595, in _dep_map
    return self.__dep_map
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2457, in __getattr__
    raise AttributeError(attr)
AttributeError: _DistInfoDistribution__dep_map

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/usr/lib/python3/dist-packages/pip/commands/install.py", line 290, in run
    requirement_set.prepare_files(finder, force_root_egg_info=self.bundle, bundle=self.bundle)
  File "/usr/lib/python3/dist-packages/pip/req.py", line 1266, in prepare_files
    req_to_install.extras):
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2401, in requires
    dm = self._dep_map
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2597, in _dep_map
    self.__dep_map = self._compute_dependencies()
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2621, in _compute_dependencies
    parsed = next(parse_requirements(distvers))
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2721, in parse_requirements
    "version spec")
  File "/usr/lib/python3/dist-packages/pkg_resources.py", line 2686, in scan_list
    raise ValueError(msg, line, "at", line[p:])
ValueError: ('Expected version spec in', 'wrapt ~=1.10', 'at', ' ~=1.10')

Storing debug log for failure in /root/.pip/pip.log
root@ts-imx6ul:/usr/lib/python3.4/site-packages/python-can#

tthatche, 2018-10-02:
So I was just trying some things.  I cloned version 2.2.0 into a local directory and tried installing to see if the older version worked.  No Luck!  So I started messing with setup.py.  Took all of the spaces and word wrap out of the dependencies.  Still No luck!  Then I took off the version information from each of the dependencies and used the "-U" option instead and now it worked.  I'll try going back to version 3.0.0 and try the same thing.

tthatche, 2018-10-02:
Issue RESOLVED!  The problem was setuptools.  See the trace back below.  

I had setuptools 5.5.1 that comes with python3-pip package.  

I upgraded with pip3 install -U setuptools to version 40.4.3 and then installed python-can successfully with the pip3 install python-can command

Is there a way to require a specific version of setuptools prior to install??

root@ts-imx6ul:~# pip3 list
chardet (2.3.0)
colorama (0.3.2)
html5lib (0.999)
pip (1.5.6)
requests (2.4.3)
setuptools (5.5.1)
six (1.8.0)
urllib3 (1.9.1)
wheel (0.24.0)
root@ts-imx6ul:~# pip3 install -U setuptools
Downloading/unpacking setuptools from https://files.pythonhosted.org/packages/96/06/c8ee69628191285ddddffb277bd5abdf769166e7a14b867c2a172f0175b1/setuptools-40.4.3-py2.py3-none-any.whl#sha256=ce4137d58b444bac11a31d4e0c1805c69d89e8ed4e91fde1999674ecc2f6f9ff
  Downloading setuptools-40.4.3-py2.py3-none-any.whl (569kB): 569kB downloaded
Installing collected packages: setuptools
  Found existing installation: setuptools 5.5.1
    Not uninstalling setuptools at /usr/lib/python3/dist-packages, owned by OS
Successfully installed setuptools
Cleaning up...
root@ts-imx6ul:~# pip3 list
chardet (2.3.0)
colorama (0.3.2)
html5lib (0.999)
pip (1.5.6)
requests (2.4.3)
setuptools (40.4.3)
six (1.8.0)
urllib3 (1.9.1)
wheel (0.24.0)
root@ts-imx6ul:~# pip3 install python-can
Downloading/unpacking python-can
  Downloading python_can-2.2.1-py2.py3-none-any.whl (136kB): 136kB downloaded
Downloading/unpacking wrapt~=1.10 (from python-can)
  Downloading wrapt-1.10.11.tar.gz
  Running setup.py (path:/tmp/pip-build-rz7zxxfd/wrapt/setup.py) egg_info for package wrapt
Installing collected packages: python-can, wrapt
  Running setup.py install for wrapt
    building 'wrapt._wrappers' extension
    arm-linux-gnueabihf-gcc -pthread -DNDEBUG -g -fwrapv -O2 -Wall -Wstrict-prototypes -g -fstack-protector-strong -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 -fPIC -I/usr/include/python3.4m -c src/wrapt/_wrappers.c -o build/temp.linux-armv7l-3.4/src/wrapt/_wrappers.o
    arm-linux-gnueabihf-gcc -pthread -shared -Wl,-O1 -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,relro -g -fstack-protector-strong -Wformat -Werror=format-security -D_FORTIFY_SOURCE=2 build/temp.linux-armv7l-3.4/src/wrapt/_wrappers.o -o build/lib.linux-armv7l-3.4/wrapt/_wrappers.cpython-34m.so
  Could not find .egg-info directory in install record for wrapt~=1.10 (from python-can)
Successfully installed python-can wrapt
Cleaning up...
root@ts-imx6ul:~#

hardbyte, 2018-10-03:
Glad to hear you've solved it. Wow [setuptools v5.5.1](https://pypi.org/project/setuptools/5.5.1/) was released 4 years ago - Aug 10, 2014

We could add a dependency on `setuptools>=40`

tthatche, 2018-10-03:
Yeah, it's interesting.  The OS is a flavor of Jessie.  But when I installed PIP with apt-get install it linked to this old version of setuptools.  

Let me know if there is any other testing you would like me to do.  Thanks to you and the contributors for keeping this going!

closed 2018-10-04 by hardbyte