hardbyte/python-can #433: Cannot receive messages with Kvaser

bessman, 2018-09-28:
Python 3.6.5, python-can 2.2.1, Kvaser CANlib 5.24, Windows 10.
bus.recv() does not return anything, even though I can see that messages are present on the bus using Kvaser CanKing. I can however send messages with bus.send() (the message shows up in CanKing).

christiansandberg, 2018-09-28:
Are the messages being sent from the same computer?

hardbyte, 2018-09-28:
Can you confirm this is an issue with the upcoming release please?

```
pip install python-can==3.0.0a1
```

tthatche, 2018-09-29:
The hardware is at work. I’ll try on Monday.

bessman, 2018-10-01:
Hmm. I can no longer reproduce this issue. I'll let you know if it reappears and I can figure out what exactly is the cause. I'll close this for now.

closed 2018-10-01 by bessman