hardbyte/python-can #457: pytest fails under python2.7

umlaeute, 2018-10-09:
while preparing the Debian packages for python-can_3.0.0, i'm having a bit of troubles with the pytest suite under.

First of all, I need to remove the following lines from `setup.cfg`, as my `pytest` (3.0.6-1 both Py2 and Py3, as installed on Debian/sid) claims that it doesn't understand those options:
~~~
[tool:pytest]
addopts = -v --timeout=300 --cov=can --cov-config=setup.cfg
~~~

After that the tests still fail (this time Python2.7 only)
~~~
.pybuild/cpython2_2.7_can/build$ python2.7 -m pytest test
========================================================= test session starts ==========================================================
platform linux2 -- Python 2.7.15+, pytest-3.6.4, py-1.6.0, pluggy-0.6.0
rootdir: /build/python-can-3.0.0, inifile:
collected 20 items / 4 errors                                                                                                          

================================================================ ERRORS ================================================================
________________________ ERROR collecting .pybuild/cpython2_2.7_can/build/test/test_detect_available_configs.py ________________________
test/test_detect_available_configs.py:18: in <module>
    from .config import IS_LINUX, IS_CI
E   ValueError: Attempted relative import in non-package
___________________________ ERROR collecting .pybuild/cpython2_2.7_can/build/test/test_message_filtering.py ____________________________
test/test_message_filtering.py:14: in <module>
    from .data.example_data import TEST_ALL_MESSAGES
E   ValueError: Attempted relative import in non-package
________________________________ ERROR collecting .pybuild/cpython2_2.7_can/build/test/test_scripts.py _________________________________
test/test_scripts.py:16: in <module>
    from .config import *
E   ValueError: Attempted relative import in non-package
___________________________ ERROR collecting .pybuild/cpython2_2.7_can/build/test/test_socketcan_helpers.py ____________________________
test/test_socketcan_helpers.py:15: in <module>
    from .config import *
E   ValueError: Attempted relative import in non-package
!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!! Interrupted: 4 errors during collection !!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!!
======================================================= 4 error in 0.11 seconds ========================================================
~~~

felixdivo, 2019-02-11:
Did you install all the dependencies from the setup.py, Like pytest-cov? Then, the options should get recognized.

felixdivo, 2019-02-12:
I just tested it on a clean machine and had a look at the CI server. I cannot reproduce you problem.

umlaeute, 2019-02-13:
the problem is with the `python-can-3.0.0.tar.gz` Release asset (as uploaded to pypi; as manually attached to the [3.0.0 release](https://github.com/hardbyte/python-can/releases/tag/3.0.0), not with the HEAD (or with the `3.0.0` snapshot, with which i had other but fixable problems.)

felixdivo, 2019-02-13:
Well, that is strange. But if HEAD works, we could simply release version 3.1 as the first version on Debian, right?

umlaeute, 2019-02-13:
to be honest, i haven't really tried `HEAD`.
For now, i've packaged the `3.0.0` git snapshot for Debian (Debian has just started to freeze for the next release; i don't know if we can push a 3.1 release at this stage).

felixdivo, 2019-02-13:
Well okay, then stay with the snapshot. I don't know when 3.1 will be released.

felixdivo, 2019-02-19:
Closing this in favor for #504.

closed 2019-02-19 by felixdivo