hardbyte/python-can #460: python-can 3.0.0 send message fail with PCAN

toolsh, 2018-10-17:
Hi All,

Recently, I upgrade python-can from 2.1.0 to 3.0.0 and get pcan messages ***sent*** fail.

I debug into pcan.py & basic.py in /can/interfaces/pcan/ and add a print in basic.py::PCANBasic::Write() ```print(MessageBuffer.DATA, MessageBuffer.LEN)```
 before
 [```res = self.__m_dllBasic.CAN_Write(Channel,byref(MessageBuffer))```](https://github.com/hardbyte/python-can/blob/958f57c5b266f31379c3687fa8e284b025d1410c/can/interfaces/pcan/basic.py#L542).
The print output shows both DATA and LEN are expected, however in receiver side (PCAN Viewer) both LEN and DATA are 0, which means all messages are recieved fail.

No this issue on python-can 2.1.0, i.e. messages are sent successfully and received correctly. (Same SW/HW environment, only python-can difference)

On the other hand, if I change ```("ID",      c_ulong)``` to ```("ID",      c_uint)``` in python-can 3.0.0
 in [```class TPCANMsg (Structure)```](https://github.com/hardbyte/python-can/blob/958f57c5b266f31379c3687fa8e284b025d1410c/can/interfaces/pcan/basic.py#L295), all mesages are sent successfully.  

I understand the comment _["ID was changed from u_uint to c_ulong, so it is compatible with the PCAN-USB Driver for macOS"](https://github.com/hardbyte/python-can/blob/develop/can/interfaces/pcan/basic.py#L294)_. I'm not sure it's been fully tested on Ubuntu 4.15.0-36-generic x86_64 with python 3.6.6.

Would any one help dig into the issue and find a solution so that I can upgrade to python-can-3.0.0 and use it on both Ubuntu and MacOS?

Any other usefully information are needed, please let me know. Thanks.

christiansandberg, 2018-10-17:
It could be that it should be c_uint32. We should check what the C-header file uses.

Lauszus, 2018-10-31:
@toolsh sorry hear to this. 

@christiansandberg I already did that and unfortunately they do not use ```uint32_t``` in C code.

The best "fix" might be just to make separate structures for Mac compatibility.

hardbyte, 2018-10-31:
I reached out to PEAK-System and got this back:

> Hello Brian, thanks for using our libraries and for the invitation to participate in the  project. Just for information, PCAN-Basic is supported already under Linux. The same python header file can be used for  Linux projects - irt can be downlaoded here:                 https://www.peak-system.com/PCAN-Basic.239.0.html?&L=1
. Our dirers doesn't support OSx so porting the API on this platform makes no sense.

Lauszus, 2018-10-31:
@hardbyte that's weird. They even link to it on their webpage: https://www.peak-system.com/MacCAN.336.0.html?&L=1.

toolsh, 2018-12-12:
Hi @Lauszus 
Is there any official fix/release plan so that python-can-3.x.x can be used on both Ubuntu and MacOS?

Lauszus, 2018-12-16:
@toolsh I hope to work on a fix this coming week.

Lauszus, 2018-12-18:
@toolsh please test https://github.com/hardbyte/python-can/pull/481.

PR #481 (2018-12-24): Create separate structs for the Mac PCAN driver
Fixes #460
commit 6ab3d8de (2018-12-24): Create separate structs for the Mac PCAN driver

Fixes #460
closed 2018-12-24 by hardbyte