hardbyte/python-can #461: Device selection for Vector CAN hardware

far3ll, 2018-10-22:
Hi all,

I'm testing python-can for automating HIL setup with lots of different hardware setups and a working CAN bus system. Robot Framework is used as test automation software with a self-written CAN API, which struggles to control two Vector VN1610 hardwares at the same time.
As far as I read in the python-can documentation, I haven't seen any parameters to choose, which hardware I want to use, if there are two or more plugged to the automation PC (beside the selectable channels). Is there a workaround or implementation for future releases planned?
Thanks in advance!

Best regards,
Aaron

christiansandberg, 2018-10-22:
You can either use the Hardware Config to assign physical channels to logical ones in an application. Alternatively you can provide a serial number to specify the hardware.

```python
bus = can.Bus(bustype='vector', app_name='python-can', channel=0)
bus = can.Bus(bustype='vector', serial=12345, channel=0)
```

far3ll, 2018-10-29:
Thank you for your reply. Assigning the hardware via serial works fine, for the physical to logical assignment, I have to read the manual and get used to it.

Cheers,
Aaron

closed 2018-10-29 by far3ll

cloudsftp, 2021-08-24:
Hey @far3ll, how did you find out the serial number of the device VN1610? I can't find anything useful online :(

far3ll, 2021-08-24:
Hi @cloudsftp,
serial number can be found by either reading it on the hardware itself (sticker on VN1610 with format XXXX-XXXX, only last 5 digits are relevant) or by accessing it in Windows Control panel (the old one, not the one from Win10). There should be a "Vector Hardware" topic (or similar name) with all connected devices and its assigned serial numbers.
Cheers!