hardbyte/python-can #463: ImportError: The Vector API has not been loaded

nagarajdesai, 2018-10-24:
I am new to python. I am trying to use this Python-can library, I get the following import error when i try to run the send_one() example code:

  File "build\bdist.win-amd64\egg\can\interface.py", line 128, in __new__
  File "build\bdist.win-amd64\egg\can\interfaces\vector\canlib.py", line 79, in __init__
ImportError: The Vector API has not been loaded

Process finished with exit code 1

Could someone please help me solve this issue?

christiansandberg, 2018-10-24:
So you see a warning too? It could not load the Vector library it seems.

nagarajdesai, 2018-10-24:
If it could not load the vector library then it should have thrown an Exception in line 43 of can/interfaces/vector/canlib.py:
[code]
# Import safely Vector API module for Travis tests
vxlapi = None
try:
    from . import vxlapi
except Exception as exc:
    LOG.warning('Could not import vxlapi: %s', exc)
[/code]

christiansandberg, 2018-10-24:
The exception is caught and issued as a warning. For me this usually shows up in the console but you may need to run Python in debug mode or configure the Python logger to print everything.

nagarajdesai, 2018-10-25:
No I don't see this warning in my log

nagarajdesai, 2018-10-25:
Actually there was a problem with my logging. I see the warning in the logs now:

"C:\Program Files\Python_27\python.exe" C:/Users/desanayb/Documents/Python-Can/python-can-release-3.0.0/python-can-release-3.0.0/can/interfaces/vector/canlib.py
WARNING:root:creating logger
WARNING:can.interfaces.vector.canlib:Could not import vxlapi: cannot import name VectorError
WARNING:root:creating logger
Traceback (most recent call last):
  File "C:/Users/desanayb/Documents/Python-Can/python-can-release-3.0.0/python-can-release-3.0.0/can/interfaces/vector/canlib.py", line 16, in <module>
    import can.interfaces.vector.vxlapi
  File "build\bdist.win-amd64\egg\can\interfaces\vector\vxlapi.py", line 298, in <module>
    xlCanReceive = _xlapi_dll.xlCanReceive
  File "C:\Program Files\Python_27\lib\ctypes\__init__.py", line 378, in __getattr__
    func = self.__getitem__(name)
  File "C:\Program Files\Python_27\lib\ctypes\__init__.py", line 383, in __getitem__
    func = self._FuncPtr((name_or_ordinal, self))
AttributeError: function 'xlCanReceive' not found

Process finished with exit code 1

christiansandberg, 2018-10-25:
Maybe try updating the vector drivers.

sjenyart, 2018-12-12:
Hello.   I am seeing this error in Linux.  As far as I can tell, the Vector drivers are only available for Windows.  Anyone have any luck getting python-can working with Vector in Linux?

sjenyart, 2018-12-12:
As far as I can tell, no.  I was wondering if I had missed something.

felixdivo, 2019-02-26:
@christiansandberg

Maybe we should add a note to the docs telling that only WIndows is supported? It's not *super* obvious at the moment.

I guess this can then be closed.

felixdivo, 2019-05-02:
@sjenyart Would you mind creating a PR with docs explaining this more clearly?

closed 2019-08-04 by felixdivo

NullPointerException00, 2019-12-23:
I run [https://can-isotp.readthedocs.io/en/latest/isotp/examples.html#basic-transmission-with-python-can](url) the Basic transmission with python-can in windows, I still got the problem that "Could not import vxlapi: Could not find module 'vxlapi'". Anyon can give me a help?

felixdivo, 2019-12-24:
Maybe you use an old version? Please check that and if not, please create a new issue for it so it does not get lost.

Merry Christmas ;)

christiansandberg, 2019-12-25:
Have you installed the drivers from Vector?

NullPointerException00, 2019-12-26:
I found the problem that I did not install the drivers! Thank you all so much.

PR #667 (2019-08-04): State "windows only" in Vector docs
Fix #463
commit 089e29cd (2019-08-04): State "windows only" in Vector docs

Fix #463