hardbyte/python-can #465: USB2CAN interface: Incomplete initialization

acolomb, 2018-10-26:
I just tried to use the USB2CAN interface under Windows (the same device works fine with SocketCAN on Linux).  It took some fiddling with the `usb2can.dll` available from 8devices, which is apparently 32-bit only and therefore only works in a 32-bit Python. I've mailed them about that issue.

However, after getting that it to work under 32-bit Python, the bus usage in my application fails.  I traced it down to the `can.interfaces.usb2can.Usb2canBus` constructor not having a `_filters` member.  That should be set in the base class constructor, `can.BusABC.__init__()`.  In contrast to the other interfaces modules, the call to the parent class constructor is missing.

I adapted the code from the `pcan` interface, but I'm not sure whether the call should happen at the beginning or at the end of the `Usb2canBus` constructor.  Let me know if you'd like a PR with the hacked version I am currently running.

rusoku, 2019-02-09:
If you would like to test 64bit usb2can.dll version I can compile it.

hardbyte, 2019-02-10:
Yes please, a working backend is better than not!

It would also be worth adding a small note to the USB2CAN backend docs to use 32-bit Python until that is sorted too.

acolomb, 2019-02-11:
@hardbyte There you go.  I'm not quite sure if it is the best idea to call the parent class initialization as last thing in the constructor, since it may overwrite some stuff that was already initialized specific to this backend?  In my testing, it worked however.

@rusoku Testing with a 64-bit `usb2can.dll` would be great, please do send it.  I see you are not directly affiliated with 8devices, but Rusoku sells a very similar looking TouCAN adapter. Is that somehow a successor product?  What is the connection between Rusoku and 8devices, besides being in the same Lithuanian town?  I'd just like to know a little more before running a DLL provided by a stranger, with system permissions under Windows ;-)

rusoku, 2019-02-11:
@acolomb usb2can converter I developed for VSCP home automation open source open hardware project long time ago.I am one of contributor CANAL inteface standard.
Later I worked for 8dev company and they has been selling my converters.At his moment I opnened my own company and I developed new converters based on Cortex M4 platform (TouCAN series etc.).Also CANAL API library are totaly rewrited.Added new function.Now it supports USB hot unplug and Microsoft extended function list.
So Windows kernel drivers no longer needed.

rusoku, 2019-02-11:
BTW usb2can firmware and canal source code lives here: https://github.com/krumboeck

acolomb, 2019-02-12:
@rusoku so would it make sense to use the CANAL library package provided on your [product page](https://www.rusoku.com/products)?  I see that it includes a 64-bit version.  However, the version number is 1.0.0 compared to the 1.0.6 version provided by 8devices for the USB2CAN.  The [source repository](https://github.com/krumboeck/usb2can_canal) has not been touched since its initial commit in 2013, but seems to have 1.0.6 (`DLL_*_VERSION`) as well.  It also says something about 1.0.14 (`CANAL_*_VERSION`), so I am now officially confused.

rusoku, 2019-02-12:
1.0.6 ir CANAL API standard version.Current CANAL standard version is 1.14.
https://grodansparadis.gitbooks.io/the-vscp-daemon/canal_interface_specification.html
1.0.0 is build version for TouCAN series converters with extended non CANAL standart API list, because standard CANAL API has limited functionality especially for new TouCAN FD converter series.

hardbyte, 2019-02-13:
Sorry didn't mean to shutdown your convo. @rusoku could I ask you to take a look at https://github.com/hardbyte/python-can/pull/245 

I'm getting the impression we have a bit of work to do with cleaning up the USB2CAN backend? Possibly including changing the name to `CANAL`?

@acolomb and @rusoku would either/both of you mind opening github issues?

CageScriptLo, 2019-02-14:
@rusoku anyway that you could make the 64bit version of usb2can.dll available?  It would be grately appreciated.

rusoku, 2019-02-14:
http://www.rusoku.com/downloads/canal_32_64_20180316.zip

CageScriptLo, 2019-02-14:
@rusoku, thank you very much.  I should have looked a little closer at your website.  Hate to push my luck and hijack the thread but I can figure out the connection string for 400kb/s on the USB2CAN device.

rusoku, 2019-02-14:
@CageScriptLo,  can you elaborate please ?

CageScriptLo, 2019-02-14:
@rusoku, trying to calculate the TS1, TS2, SJW, and BRP for the custom speed.

rusoku, 2019-02-14:
@CageScriptLo, 
should be: tseg1=12; tseg2=2; brp=5; sjw=1;
init string format for usb2can:  serial_number;0;tseg2;tseg1;sjw;brp

acolomb, 2019-02-19:
@rusoku I am currently not working on my project involving the USB2CAN adapter very regularly.  When getting back to that, I will test with your provided 64-bit DLL, though.  Thank you very much.

If you still have contact to the team at 8devices, you could offer it to them as well, for their product download area.  I've never received a reply to my request for the DLL and the whole CANAL development is hard to grasp for an outsider like me.  Especially with your (TouCAN) CANAL now being something different with the same name and jumping back to version 1.0.0.

rusoku, 2019-02-20:
@acolomb new usb2can device firmwares, windows kernels driver, canal versions was provided to them.They never added it to download section.

acolomb, 2019-03-10:
After testing the stuff in #511, this seems to work as expected.  Where to get the "official" matching CANAL DLLs from is still a source of confusion, but not an issue with this project and not related to the problem described here. Hence closing this issue now.

PR #501 (2019-02-13): Fully initialize usb2can Bus interface object.
Add the missing parent class initialization to fix #465.

Tested under a 32-bit Python3 environment in Windows 7.
closed 2019-03-10 by acolomb