hardbyte/python-can #473: Cannot write to file

bessman, 2018-11-20:
```
pr = can.Printer()
csvw = can.CSVWriter('output.csv')
n = can.Notifier(bus, [pr, csvw])
```
Expected result: output.csv should contain the same messages which are printed to stdout by pr.
Actual result: pr prints messages to stdout, but output.csv is empty.

With pr = can.Printer(file='output.csv') instead nothing is written to either stdout or output.csv.

Version 3.0.0, Windows 10.

travlytle, 2018-11-20:
I have the same issue.  python 32bit 3.7.1 windows 7 using the following code:

import time
import can

```
bus = can.interface.Bus(bustype='pcan', channel='PCAN_USBBUS1', bitrate=500000)

can_filters = [{"can_id": 0x500, "can_mask": 0xFFF, "extended": False}]
bus.set_filters(can_filters)

# iterate over received messages
for msg in bus:
    print("{:X}: {}".format(msg.arbitration_id, msg.data.hex()))

    # or use an asynchronous notifier
    notifier = can.Notifier(bus, [can.Logger("recorded.log"), can.Printer()])

```

my output to screen looks like this:
Timestamp:    54593.925514        ID: 0500    S          DLC: 8    01 06 6e 00 33 34 06 01
Timestamp:    54593.945565        ID: 0500    S          DLC: 8    00 20 d0 68 f1 34 a4 b5
Timestamp:    54594.326452        ID: 0500    S          DLC: 8    04 01 cb 00 00 00 00 00
Timestamp:    54594.905405        ID: 0500    S          DLC: 8    02 18 31 1a 7f 0c e2 04
Timestamp:    54594.925328        ID: 0500    S          DLC: 8    01 06 6f 00 33 34 06 01
Timestamp:    54594.945550        ID: 0500    S          DLC: 8    00 20 d0 68 f1 34 a4 b4
Timestamp:    54595.326351        ID: 0500    S          DLC: 8    04 01 d3 00 00 00 00 00
Timestamp:    54595.525497        ID: 0500    S          DLC: 8    02 00 00 00 00 00 00 00
Timestamp:    54595.545454        ID: 0500    S          DLC: 8    01 e4 05 00 33 38 06 01
Timestamp:    54595.565463        ID: 0500    S          DLC: 8    00 a1 67 9b 0e 6c 54 b2
500: 0218271a750cd904
Timestamp:    54595.925445        ID: 0500    S          DLC: 8    01 06 70 00 33 34 06 01
Timestamp:    54595.945496        ID: 0500    S          DLC: 8    00 20 d0 68 f1 34 a4 b1
Timestamp:    54596.225358        ID: 0500    S          DLC: 8    02 00 00 00 00 00 00 00
500: 01e4050033390601
Timestamp:    54596.265333        ID: 0500    S          DLC: 8    00 f6 fb 69 22 91 b0 b4
Timestamp:    54596.306288        ID: 0500    S          DLC: 8    02 00 00 00 00 00 00 00
Timestamp:    54596.326297        ID: 0500    S          DLC: 8    01 e4 05 00 33 37 06 01
Timestamp:    54596.345281        ID: 0500    S          DLC: 8    00 51 55 9b 0e 6c 54 b7
Timestamp:    54596.365332        ID: 0500    S          DLC: 8    04 01 d1 00 00 00 00 00
Timestamp:    54596.905239        ID: 0500    S          DLC: 8    02 18 2c 1a 7d 0c e0 04
Timestamp:    54596.925247        ID: 0500    S          DLC: 8    01 06 71 00 33 34 06 01
Timestamp:    54596.945384        ID: 0500    S          DLC: 8    00 20 d0 68 f1 34 a4 b5
Timestamp:    54597.325246        ID: 0500    S          DLC: 8    04 01 cf 00 00 00 00 00
Timestamp:    54597.905254        ID: 0500    S          DLC: 8    02 18 31 1a 7f 0c e2 04

My file is created but is empty.

prabueac, 2018-11-21:
I am facing same problem , the log file remains empty ,,the output message displayed on screen

travlytle, 2018-11-21:
I thought maybe the issue was with pcan, but I tested on a raspberry pi with socketcan and still have the same issue.

prabueac, 2018-11-21:
if its issue with PCAN , you can't see the CAN message on screen ? as displayed above

travlytle, 2018-11-21:
The behavior is the same between pcan and socketcan.  Both don't record to a file and will show the same data on the screen, as I posted above.

prabueac, 2018-11-21:
i just tested with command line with following args
python.exe -m  can.logger -f test_log.csv -c PCAN_USBBUS1 -i pcan -b 125000

Output on test_log.csv   file 

timestamp | arbitration_id | extended | remote | error | dlc | data |  
-- | -- | -- | -- | -- | -- | -- | --
127490 | 0x5003157 | 1 | 0 | 0 | 8 | GxUxcQdxVwA=
127491 | 0x5054400 | 1 | 0 | 0 | 8 | FURxBpdYAAA=
127492 | 0x5003157 | 1 | 0 | 0 | 8 | GxUxcQdxVwA=
127494.1 | 0x5003157 | 1 | 0 | 0 | 8 | GxUxcQdxVwA=
127494.9 | 0x5044400 | 1 | 0 | 0 | 8 | FURxBolYAAA=
127496.2 | 0x5003157 | 1 | 0 | 0 | 8 | GxUxcQdxVwA=
127498.3 | 0x5003157 | 1 | 0 | 0 | 8 | GxUxcQdxVwA=
127500 | 0x5034400 | 1 | 0 | 0 | 8 | FUNxAmlwAAA=
127500.4 | 0x5003157 | 1 | 0 | 0 | 8 | GxUxcQdxVwA=
127502.5 | 0x5003157 | 1 | 0 | 0 | 8 | GxUxcQdxVwA=
127502.9 | 0x5024400 | 1 | 0 | 0 | 8 | FURxBpEJAAA=
127504.3 | 0x5014400 | 1 | 0 | 0 | 8 | FTFxB3FXAAA=
127504.5 | 0x5003157 | 1 | 0 | 0 | 8 | GxUxcQdxVwA=
127506.6 | 0x5003157 | 1 | 0 | 0 | 8 | GxUxcQdxVwA=
127506.6 | 0x5054400 | 1 | 0 | 0 | 8 | FURxBpdYAAA=
127508.7 | 0x5003157 | 1 | 0 | 0 | 8 | GxUxcQdxVwA=
127510.5 | 0x5044400 | 1 | 0 | 0 | 8 | FURxBolYAAA=
127510.8 | 0x5003157 | 1 | 0 | 0 | 8 | GxUxcQdxVwA=
127512.9 | 0x5003157 | 1 | 0 | 0 | 8 | GxUxcQdxVwA=
127514.9 | 0x5003157 | 1 | 0 | 0 | 8 | GxUxcQdxVwA=
127515.6 | 0x5034400 | 1 | 0 | 0 | 8 | FUNxAmlwAAA=
127517 | 0x5003157 | 1 | 0 | 0 | 8 | GxUxcQdxVwA=
127518.5 | 0x5024400 | 1 | 0 | 0 | 8 | FURxBpEJAAA=
127519.1 | 0x5003157 | 1 | 0 | 0 | 8 | GxUxcQdxVwA=
127520 | 0x5014400 | 1 | 0 | 0 | 8 | FTFxB3FXAAA=
127521.2 | 0x5003157 | 1 | 0 | 0 | 8 | GxUxcQdxVwA=
127522.2 | 0x5054400 | 1 | 0 | 0 | 8 | FURxBpdYAAA=
127523.3 | 0x5003157 | 1 | 0 | 0 | 8 | GxUxcQdxVwA=

[image]