hardbyte/python-can #479: Message data incorrect after Bus.recv

sjenyart, 2018-12-13:
Behavior seems to be similar to what is described in #129 

Screenshot below shows the Peak System PCAN-USB transmitting three frames (id=0x194) to wake up the remote node.  The messages below use the receive_all() example code to print incoming messages.  As you can see in the trace on the left, the remote node is transmitting a heartbeat message (id=0x702) with a single byte in the payload, 0x7f.

The received message is nonsensical.  This is in a Mint 19 VBox.  Peak Linux driver 8.7.0,  PCAN Basic 4.2.2.

https://imgur.com/QuVczxO

Let me know if I can help.  Thanks.

felixdivo, 2019-04-27:
@sjenyart Were you able to find the cause of this bug?

felixdivo, 2021-04-23:
Cannot reproduce.

closed 2021-04-23 by felixdivo