hardbyte/python-can #482: Socketcan: No buffer space available when can logging level != debug.

sjenyart, 2018-12-19:
I am using CANopen for Python, which is utilizing socket can for transmissions.  I would like to turn off these messages:

> DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0000    S                DLC:  2    01 00

If I change this line in my code: 
`can.util.set_logging_level(level_name='**debug**')`

to:
`can.util.set_logging_level(level_name='**info**')`

I receive the following trace:

> Traceback (most recent call last):
  File "/builds/.../ci/hil-test/lib/hil_test.py", line 76, in run
    case.run(self.status)
  File "/builds/.../ci/hil-test/tests/validate_canopen.py", line 110, in run
    network.scanner.search()
  File "/usr/local/lib/python3.5/dist-packages/canopen/network.py", line 380, in search
    self.network.send_message(0x600 + node_id, sdo_req)
  File "/usr/local/lib/python3.5/dist-packages/canopen/network.py", line 201, in send_message
    self.bus.send(msg)
  File "/usr/local/lib/python3.5/dist-packages/can/interfaces/socketcan/socketcan.py", line 547, in send
    sent = self._send_once(data, msg.channel)
  File "/usr/local/lib/python3.5/dist-packages/can/interfaces/socketcan/socketcan.py", line 570, in _send_once
    raise can.CanError("Failed to transmit: %s" % exc)
can.CanError: Failed to transmit: [Errno 105] No buffer space available

Looking through socketcan.py, it is not perfectly clear to me why we're seeing this failure.  I would assume that these lines would simply result in nothing happening if the logging level is set to a higher value:

      
`log.debug("We've been asked to write a message to the bus")`
`logger_tx = log.getChild("tx")`
`logger_tx.debug("sending: %s", msg)`

hardbyte, 2018-12-23:
Sorry to hear you're running into trouble - I tried to replicate it (using virtual socketcan) with the `examples/cyclic.py` script but don't have any problems:

With logging level set to debug:
```
Carrying out cyclic tests with socketcan interface
DEBUG:can.util:Logging set to debug
Starting to send a message every 200ms for 2s
DEBUG:can:can config: {'bitrate': 500000, 'interface': 'socketcan', 'channel': 'vcan0'}
INFO:can.interfaces.socketcan.socketcan:Created a socket
DEBUG:can.interfaces.socketcan.socketcan:Binding socket to channel=vcan0
DEBUG:can.interfaces.socketcan.socketcan:Bound socket.
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0000    S                DLC:  6    00 00 00 00 00 00
DEBUG:can.interfaces.socketcan.socketcan:Sending BCM command
DEBUG:can.interfaces.socketcan.socketcan:Stopping periodic task
DEBUG:can.interfaces.socketcan.socketcan:We've been asked to write a message to the bus
DEBUG:can.interfaces.socketcan.socketcan.tx:sending: Timestamp:        0.000000        ID: 0000    S                DLC:  6    00 00 00 00 00 00
DEBUG:can.interfaces.socketcan.socketcan:sending an extended id type message
DEBUG:can.interfaces.socketcan.socketcan:sending an extended id type message
DEBUG:can.interfaces.socketcan.socketcan:Sending BCM command
stopped cyclic send
Starting to send a message every 200ms for 1s
Cyclic send should have stopped as duration expired
Starting to send a message every 200ms. Initial data is ones
DEBUG:can.interfaces.socketcan.socketcan:sending an extended id type message
DEBUG:can.interfaces.socketcan.socketcan:sending an extended id type message
DEBUG:can.interfaces.socketcan.socketcan:Sending BCM command
Changing data of running task to begin with 99
DEBUG:can.interfaces.socketcan.socketcan:sending an extended id type message
stopped cyclic send
Changing data of stopped task to single ff byte
DEBUG:can.interfaces.socketcan.socketcan:Stopping periodic task
DEBUG:can.interfaces.socketcan.socketcan:sending an extended id type message
starting again
DEBUG:can.interfaces.socketcan.socketcan:sending an extended id type message
DEBUG:can.interfaces.socketcan.socketcan:sending an extended id type message
DEBUG:can.interfaces.socketcan.socketcan:Sending BCM command
done
DEBUG:can.interfaces.socketcan.socketcan:Stopping periodic task
DEBUG:can.interfaces.socketcan.socketcan:Closing bcm socket for channel vcan0
DEBUG:can.interfaces.socketcan.socketcan:Closing raw can socket
```

With logging level set to info:

```
Carrying out cyclic tests with socketcan interface
INFO:can.interfaces.socketcan.socketcan:Created a socket
Starting to send a message every 200ms for 2s
stopped cyclic send
Starting to send a message every 200ms for 1s
Cyclic send should have stopped as duration expired
Starting to send a message every 200ms. Initial data is ones
Changing data of running task to begin with 99
stopped cyclic send
Changing data of stopped task to single ff byte
starting again
done
```

christiansandberg, 2018-12-23:
Could it just be that when debug info is not printed, the application has enough resources to send more messages than the bus can handle?

felixdivo, 2019-02-27:
I'm closing this since we cannot reproduce this. Please comment if you have further questions.

closed 2019-02-27 by felixdivo