hardbyte/python-can #489: Property is not implemented. on Can1

patrickpoirier51, 2019-01-13:
Try to get my BeagleBone to talk CAN 
I can send and read (cansend  - candump) to my can device setup in loopback
sudo ip link set can1 up type can bitrate 1000000 loopback on

I modified the script receive_all.py  like this:
bus = can.interface.Bus(bustype='socketcan', channel='can1', bitrate=10000000)

python3 receive_all.py
Traceback (most recent call last):
  File "receive_all.py", line 29, in <module>
    receive_all()
  File "receive_all.py", line 16, in receive_all
    bus.state = BusState.ACTIVE
  File "/usr/local/lib/python3.5/dist-packages/python_can-3.0.0-py3.5.egg/can/bus.py", line 382, in state
NotImplementedError: Property is not implemented.

Pleas advise

Lauszus, 2019-01-13:
Seems like you have one zero too many in ```bitrate=10000000```. I.e. you are trying to set the bitrate to 10 million and not 1 million.

patrickpoirier51, 2019-01-13:
OMG !! You are right ... I reduced to 1000000 with same issue (even reduced to 250000)

Traceback (most recent call last):
  File "receive_all.py", line 29, in <module>
    receive_all()
  File "receive_all.py", line 16, in receive_all
    bus.state = BusState.ACTIVE
  File "/usr/local/lib/python3.5/dist-packages/python_can-3.0.0-py3.5.egg/can/bus.py", line 382, in state
NotImplementedError: Property is not implemented.

patrickpoirier51, 2019-01-13:
I see on line 382 that it sends state parameters:
        :param new_state: BusState.ACTIVE, BusState.PASSIVE or BusState.ERROR

here my can1 status:
debian@beaglebone:~/python-can/examples$ ip -details link show can1
3: can1: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UNKNOWN mode DEFAULT group default qlen 10
    link/can  promiscuity 0
    can state ERROR-PASSIVE (berr-counter tx 128 rx 0) restart-ms 0
          bitrate 1000000 sample-point 0.750
          tq 83 prop-seg 4 phase-seg1 4 phase-seg2 3 sjw 1
          c_can: tseg1 2..16 tseg2 1..8 sjw 1..4 brp 1..1024 brp-inc 1
          clock 24000000numtxqueues 1 gso_max_size 65536 gso_max_segs 65535

christiansandberg, 2019-01-13:
You can’t change the state in SocketCAN. At least not in user space as that would affect all users.

patrickpoirier51, 2019-01-13:
So I guess this tool is not for suited for SocketCan setup ?
If it's the case,is there a clear indication in the wiki ?

Best Regards

closed 2019-01-13 by patrickpoirier51