hardbyte/python-can #495: Best way to send messages with different data every seconds

aivs, 2019-02-02:
I developed digital dashboard for my car. Some data I want to request every 1 seconds (temperature engine), some data 10 times in second (RPM).
I need several tasks at one time, do I need to use threads or is there another way?
My code now uses only 1 task to get doors status.

```
GET_OIL_TEMPERATURE_COMMAND = 0x2F
GET_DOORS_COMMAND = 0x0D
GET_OUTDOOR_TEMPERATURE = 0x0C
GET_INDOOR_TEMPERATURE = 0x13
GET_ENGINE_TEMPERATURE = 0x05

import can
bus = can.interface.Bus(channel='can0', bustype='socketcan')
msg_get_oil_temperature = 		can.Message(arbitration_id=0x714, data=[0x03, 0x22, 0x20, GET_OIL_TEMPERATURE_COMMAND, 0x55, 0x55, 0x55, 0x55], extended_id=False)
msg_get_doors =					can.Message(arbitration_id=0x714, data=[0x03, 0x22, 0x22, GET_DOORS_COMMAND, 0x55, 0x55, 0x55, 0x55], extended_id=False)
msg_get_outdoor_temperature = 	can.Message(arbitration_id=0x714, data=[0x03, 0x22, 0x22, GET_OUTDOOR_TEMPERATURE, 0x55, 0x55, 0x55, 0x55], extended_id=False)
msg_get_indoor_temperature = 	can.Message(arbitration_id=0x746, data=[0x03, 0x22, 0x26, GET_INDOOR_TEMPERATURE, 0x55, 0x55, 0x55, 0x55], extended_id=False)
msg_get_engine_temperature = 	can.Message(arbitration_id=0x714, data=[0x03, 0x22, 0xF4, GET_ENGINE_TEMPERATURE, 0x55, 0x55, 0x55, 0x55], extended_id=False)

if __name__ == "__main__":
	# Start periodec send
	task = bus.send_periodic(msg_get_oil_temperature, 1)

	# Parse data
	for message in bus:
		if message.arbitration_id == 0x77E and message.data[3] == GET_OIL_TEMPERATURE_COMMAND:
			print('Температура масла = ',message.data[4]-58,'°C')

		if message.arbitration_id == 0x77E and message.data[3] == GET_DOORS_COMMAND:
			#01010101 = 0x55 (all doors closed)
			if message.data[4]&1 != 0:
				print('Водительская дверь закрыта')
			else:
				print('Водительская дверь открыта')
			if message.data[4]&4 != 0:
				print('Пассажирская дверь закрыта')
			else:
				print('Пассажирская дверь открыта')
			if message.data[4]&16 != 0:
				print('Задняя левая дверь закрыта')
			else:
				print('Задняя левая дверь открыта')
			if message.data[4]&64 != 0:
				print('Задняя правая дверь закрыта')
			else:
				print('Задняя правая дверь открыта')
```

What is the best way to send messages with different data every seconds?

hardbyte, 2019-02-02:
Depending on your backend using the same arbitration id with different periodic tasks might not work.

I'm most familiar with `socketcan` which distinguishes the tasks by their unique can_id. In that case there are a few options:
1) If the request rates aren't too extreme you could probably do it all in native Python. Have a look at our [thread based cycle send task](https://github.com/hardbyte/python-can/blob/develop/can/broadcastmanager.py#L105).
2) You could [create another BCM socket](https://github.com/hardbyte/python-can/blob/develop/can/interfaces/socketcan/socketcan.py#L184) for each task (or modify the python-can library [here](https://github.com/hardbyte/python-can/blob/develop/can/interfaces/socketcan/socketcan.py#L602)) 
3) you could do all the requesting manually in the application layer without using the periodic sending capability. If you haven't seen X in a given time send the appropriate request message...

TravisJoe, 2019-02-03:
Hello,

With the frequency you want to send I agree that you might want to just do it all in native Python until you know you need more performance. I would recommending threading, multiprocessing solutions in your main application has a main loop that takes more than about 10ms or is unpredictable. It also depends on how precise of timing you need to send these messages. @hardbyte has good options if you want to go down this path. In the past i have also creating a separate process just managing CAN. 

One thing I will always suggest is that if you are not logging and gathering everything, using CAN filters will make life much easier. This takes a huge burden from your application code whether you are using threads or not.

One pattern I like is using dictionaries similar to switch/case statements instead of a large amount of if statements. It also allows for easier abstraction and readability. This not clean or perfect, just examples of what I mean.

``` python
parse_cmd = {
    0x2F: GET_OIL_TEMPERATURE_COMMAND,
    0x0D: GET_DOORS_COMMAND,
    0x0C: GET_OUTDOOR_TEMPERATURE,
    0x13: GET_INDOOR_TEMPERATURE,
    0x05: GET_ENGINE_TEMPERATURE,
}

if message.arbitration_id == 0x77E:
    cmd = message.data[3]
    parse_cmd[cmd](message.data)

def GET_DOORS_COMMAND(data):
    if data&1 > 0:
        print('Водительская дверь закрыта')
    else:
        print('Водительская дверь открыта')
```

Because your frequency of messages relative to CAN bus and a simple application is not that fast you can use simple differences in system time. It sounds like 100ms intervals is the fastest you are looking for which should be plenty easy if you are fine with a little variation due to program and OS variables.

``` python
import time

rpm_last_sent = time.time()
rpm_delta = 100 # change in time in ms

if time.time - rpm_last_sent >= rpm_delta:
    # Send CAN message here
    rpm_last_sent = time.time()
```

I know these are simple, but not a bad place to start when you are not sure how deep you need to go. I hope this is helpful as I am quickly writing this and need to get back to work.

aivs, 2019-02-06:
Big Thanks!
Now I'm using threading and it works. For RPM's and Speed I'll add new thread.

```python
#!/usr/bin/env python3
# coding: utf-8
import can
from threading import Thread
import time

messageCommands = {
'GET_OIL_TEMPERATURE_COMMAND' : 0x2F,
'GET_DOORS_COMMAND' : 0x0D,
'GET_OUTDOOR_TEMPERATURE' : 0x0C,
'GET_INDOOR_TEMPERATURE' : 0x13,
'GET_ENGINE_TEMPERATURE' : 0x05
}

bus = can.interface.Bus(channel='can0', bustype='socketcan')

canCommands = [
	can.Message(arbitration_id=0x714, data=[0x03, 0x22, 0x20, messageCommands['GET_OIL_TEMPERATURE_COMMAND'], 0x55, 0x55, 0x55, 0x55], extended_id=False),
	can.Message(arbitration_id=0x714, data=[0x03, 0x22, 0x22, messageCommands['GET_DOORS_COMMAND'], 0x55, 0x55, 0x55, 0x55], extended_id=False),
	can.Message(arbitration_id=0x714, data=[0x03, 0x22, 0x22, messageCommands['GET_OUTDOOR_TEMPERATURE'], 0x55, 0x55, 0x55, 0x55], extended_id=False),
	can.Message(arbitration_id=0x746, data=[0x03, 0x22, 0x26, messageCommands['GET_INDOOR_TEMPERATURE'], 0x55, 0x55, 0x55, 0x55], extended_id=False),
	can.Message(arbitration_id=0x714, data=[0x03, 0x22, 0xF4, messageCommands['GET_ENGINE_TEMPERATURE'], 0x55, 0x55, 0x55, 0x55], extended_id=False)
	]

class requestsLoop(Thread):
	def __init__(self):
		Thread.__init__(self)
		self.daemon = True
		self.start()
	def run(self):
		while True:
			for command in canCommands:
				bus.send(command)
				time.sleep(0.1)

if __name__ == "__main__":
	# Send requests
	requestsLoop()

	# Parse data
	for message in bus:
		if message.arbitration_id == 0x77E and message.data[3] == messageCommands['GET_OIL_TEMPERATURE_COMMAND']:

```

mindfuucker, 2019-02-06:
That looks great, you could also use a listener or even multiple listeners. 

messageCommands = {
'GET_OIL_TEMPERATURE_COMMAND' : [0x714, 0x2F] ,
'GET_DOORS_COMMAND' :  [0x714, 0x0D],
'GET_OUTDOOR_TEMPERATURE' :  [0x714, 0x0C],
'GET_INDOOR_TEMPERATURE' :  [0x746, 0x13],
'GET_ENGINE_TEMPERATURE' :  [0x714,  0x05]
}

class SpecificListener(can.Listener):
    def __init__(self, arbitration_id, datafilter):
        self.arbitration_id = arbitration_id
        self.datafilter = datafilter

    def passedfilter(arbitration_id, data):
        return self.arbitration_id == arbitration_id and data[3] == self.datafilter

    def on_message_received(self, my_message):
        if passedfilter(my_message.arbitration_id, my_message.data):
            # Do something with the message
            pass

if __name__ == "__main__":
    # Send requests
    requestsLoop()
    notifier = can.Notifier(bus, [], timeout=60)
    for messageparameters in messageCommands.values()
        listener = SpecificListener(messageparameters[0], messageparameters[1])
        notifier.add_listener(listener)

closed 2019-02-24 by hardbyte