hardbyte/python-can #496: Network is down

Tanrikulu, 2019-02-03:
Hello all,

here is a test-code:

`import can` 
`bus = can.Bus(interface='socketcan', channel='can0', receive_own_messages=True)` 
`message= can.Message(arbitration_id=123, is_extended_id=False, data=[0x11, 0x22, 0x33])` 
`bus.send(message, timeout=0.2)`

When I run this, I get an error-message like 'network is down'. But after typing 'sudo ip link set up can0' in shell, the test-code works.

Does anyone know how to set up the can network in a python script?

Best regards
Muhammed Tahir Tanrikulu

TravisJoe, 2019-02-03:
Not sure if this is the best way, because it has been awhile since I used this code, but I usually have a hardware/os setup function that does calls setup CAN outside of the app. Not the best example because I would write it differently now, but show at least one method of doing this all from the script. 

```python
	def _hardwareConfig(self, channel):
		'''
		Pass channel string (example 'can0') to configure OS level drivers and interface.
		'''

		self.log.info('CAN hardware OS drivers and config for CAN0')

		#General CAN bus setup on the linux kernel
		subprocess.call(['modprobe', 'can']) #
		subprocess.call(['modprobe', 'can-dev']) #
		subprocess.call(['modprobe', 'can-raw']) #basic raw CAN packets
		subprocess.call(['modprobe', 'can-bcm']) #Broadcast manager

		subprocess.call(['ip', 'link', 'set', channel, 'type', 'can', 'bitrate', '500000' , 'restart-ms' , '1500'])
		subprocess.call(['ip', 'link', 'set', 'up', channel])
```

Tanrikulu, 2019-02-05:
Thank you very much for your feedback. I will try it.

felixdivo, 2019-02-12:
That is how I do it as well. The only problem is that root/sudo permissions may be required for calling `ip link`.

felixdivo, 2019-02-12:
This is documented [here](https://python-can.readthedocs.io/en/stable/interfaces/socketcan.html) btw.

closed 2019-02-12 by felixdivo

pergolafabio, 2021-02-13:
Hey @felixdivo 

I am new to can, i want to setup canbus too with python or bash... But how, did you solve the sudo issue ? How do you pass the password?

Thnx

felixdivo, 2021-02-14:
Well, I can't give really you a secure recommendation. But one that works can be found here: https://unix.stackexchange.com/a/369868/240867. You could set it via [*visudo*](https://www.sudo.ws/man/1.8.13/visudo.man.html).