hardbyte/python-can #500: PICAN2 with python-can

bhongtip, 2019-02-11:
I'm using Raspberry pi with PICAN2 to communicate with ECU. I want to
1. send message id 0x285 : 00 FF 02 01 03 06 85 66 every 10 msec
2. the ECU send message id 0x384 every 100 msec.  I want to read message 0x384 when I press button on my program.

the 1st task is done by using send_periodically

However, when I use bus.recv(timeout=0.1), I don't get the latest data.  I connect this system with another can tool and monitor it.  with this monitor can tool, the 0x384 CAN message is already changed but what I read from bus.recv() remain at the old value.   How can I get the latest value ?

Thank you in advance.

ihebs3ideni, 2019-02-12:
have you tried using `for msg in bus:
print(msg)`?

felixdivo, 2019-02-25:
As we do not have steps to reproduce this, I am closing this. Please simply comment if you need further help and can provide some code to narrow down the problem.

closed 2019-02-25 by felixdivo