hardbyte/python-can #504: Add 3.1.0 to PyPI?

albireox, 2019-02-13:
Right now the PyPI repo only has 3.1.0rc1, which requires the `--pre` flag to be installed. I have a project that requires the changes made between 3.0.0 and 3.1.0 and I'd like to be able to use `python-can` directly from pip. Is there a reason to not release the stable version in PyPI?

felixdivo, 2019-02-19:
Here is the relevant milestone [*3.1 Release*](https://github.com/hardbyte/python-can/milestone/6). The only interesting issue stopping us from releasing it is whether or not we want to include #356 into this version. The serial interface stuff is unlikely to be rebased/updated anyway.

@hardbyte What do you think? Or should we generally do #356 in a major version bump since it might break things? (cc @christiansandberg)

hardbyte, 2019-02-20:
Yeah we can release now and move those remaining issues to a new milestone. @felixdivo do you (or anyone else) want to edit the change log based on https://github.com/hardbyte/python-can/compare/3.1.0-rc1...develop and create a github release for 3.1.0?

felixdivo, 2019-02-20:
Yeah, I can do that this weekend.

hardbyte, 2019-02-24:
You can now find python-can==3.1.0 on pypi - https://pypi.org/project/python-can/3.1.0/

closed 2019-02-24 by hardbyte

jluttine, 2019-02-24:
Great! Just a few questions about 3.1.0:

- Why the PyPI source tarball `test` directory is lacking a lot of files that are in the repo `test` directory? I'd like to run the tests as one step in a build process for a Linux distro.

- Could you tag the release commit in the git repo too?

felixdivo, 2019-02-24:
> Could you tag the release commit in the git repo too?

@hardbyte Has just done that.

hardbyte, 2019-02-24:
Thanks! I did tag the release (in master), and have opened an issue regarding the missing test scripts. Out of interest how do you run them from pypi?

albireox, 2019-02-24:
Not sure if this is relevant but while installing it I got

```
Collecting windows-curses (from python-can)
  Could not find a version that satisfies the requirement windows-curses (from python-can) (from versions: )
No matching distribution found for windows-curses (from python-can)
```

hardbyte, 2019-02-24:
Tracking in #520