hardbyte/python-can #505: RecursionError When Using Message from BLFReader

TheHankDu, 2019-02-14:
Hi All,

I am trying to run BLFReader using multiprocessing.pool.imap to speed up the parsing process(instead of a for message in BLFReader loop) and then decode the message. However, it will always enter a recursion error in Message __getattr__. Am I doing anything wrong when handling BLFReader? If so, how would I improve the process or just use the for loop?

Error Code: 

  File ".local/lib/python3.6/site-packages/can/message.py", line 57, in __getattr__
    return self._dict[key]
RecursionError: maximum recursion depth exceeded

Thank you very much for your time and hard working!

HD

hardbyte, 2019-02-14:
Yeah I don't think multiprocessing parsing will work without some serious reworking of the `BLFReader`.

It is going through a single binary file and yielding `Message` instances. Have you confirmed that the parsing the bottleneck?

christiansandberg, 2019-02-14:
You will either have to rewrite the parser yourself or if you have a lot of different log files to process you can process them in parallel.

closed 2019-02-14 by hardbyte