hardbyte/python-can #506: add MF4 io writer

danielhrisca, 2019-02-15:
I would like to implement support for exporting to asam MF4 files (https://www.asam.net/standards/detail/mdf/). That would require my library as dependency ([asammdf](https://www.asam.net/standards/detail/mdf/). Is this acceptable?

christiansandberg, 2019-02-15:
Wouldn’t that also require decoding raw messages to actual signal values?

danielhrisca, 2019-02-15:
That is not mandatory. 

The raw messages can be dumped and there are two recording modes that I would like to implement

1. unsorted mode : all messages are placed in the same channel group 
2. sorted mode: each message ID in dedicated channel groups

A database can be embedded in the MF4 file (dbc or arxml) and it can be used to later extract signal values. It is also possible to have anonymous logging if no database is attached.

christiansandberg, 2019-02-16:
I think it sounds like a good idea. But maybe add the library as an [extra](https://setuptools.readthedocs.io/en/latest/setuptools.html#id16) dependency.

felixdivo, 2019-02-19:
This sounds like a good idea! As not too many will be using this (i guess) we should add this as an extra. It would probably be a good time to add the allowed extras to a section called *Installation* or something like that in the Readme.

felixdivo, 2019-04-07:
@danielhrisca Do you need any help with this? I could also write a small tutorial on how to add new `Reader`s/`Writer`s. Would that help?

danielhrisca, 2019-04-07:
It would help, thank you. Unfortunately I did not have time to tackle this get.

felixdivo, 2019-04-11:
I wrote some text on that. See: https://github.com/hardbyte/python-can/pull/548/files#diff-200667e0b2eaf351c0585ba21fb7a144R70. Hope it helps. Please ask if questions remain,

danielhrisca, 2019-04-18:
I have a question regarding the message timestamp: does it always use the local time?

felixdivo, 2019-04-18:
I think usually [`time.time()`](https://docs.python.org/3/library/time.html#time.time) is used.

danielhrisca, 2019-04-22:
@felixdivo can you have a look at #554 ?

felixdivo, 2019-04-22:
Sure

PR #554 (2019-09-04): Add support for the mf4 io format
implements #506

(Closes #506)
PR #1289 (2023-04-03): Mf4 io
I rebased https://github.com/hardbyte/python-can/pull/554/

Closes #506
closed 2023-04-03 by zariiii9003