hardbyte/python-can #520: Broken Install - Release 3.1.0

gpotter2, 2019-02-24:
Hi !

We're having a global failure in all our tests that depend on `python-can`, since the release of 3.1.0 a few hours ago (we use `python-can` as part of our CAN implementation on [secdev/scapy](https://github.com/secdev/scapy)). The package installation fail during PyPi setup.

I was able to reproduce locally:

```shell
gpotter@vmg:~/github/scapy$ pip install python-can
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Collecting python-can
  Using cached https://files.pythonhosted.org/packages/02/01/133f113b9914b26891f0c13be61823872f9216ab0fa1859f6f1b7a15c122/python_can-3.1.0-py2.py3-none-any.whl
Collecting wrapt~=1.10 (from python-can)
  Using cached https://files.pythonhosted.org/packages/67/b2/0f71ca90b0ade7fad27e3d20327c996c6252a2ffe88f50a95bba7434eda9/wrapt-1.11.1.tar.gz
Collecting windows-curses (from python-can)
  Could not find a version that satisfies the requirement windows-curses (from python-can) (from versions: )
No matching distribution found for windows-curses (from python-can)
```

Some background data:
```
gpotter@vmg:~/github/scapy$ uname -a
Linux vmg 4.15.0-45-generic #48-Ubuntu SMP Tue Jan 29 16:27:25 UTC 2019 i686 i686 i686 GNU/Linux
gpotter@vmg:~/github/scapy$ lsb_release -a
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.2 LTS
Release:        18.04
Codename:       bionic
```

I tried with `PyPI` 9.0.1 (installed by default) and 19.0.3 (latest), with the same results.

I'll be trying out stuff in the meanwhile,

Thanks for reading !

felixdivo, 2019-02-24:
`windows-curses` is still on PyPi: https://pypi.org/project/windows-curses/

felixdivo, 2019-02-24:
And it works on our Windows builds: https://ci.appveyor.com/project/hardbyte/python-can/builds/22606096/job/81689cpva238ul9f#L78. And our linux builds do not install it https://ci.appveyor.com/project/hardbyte/python-can/builds/22606096/job/81689cpva238ul9f#L78. Why do your Unix builds even install that?

gpotter2, 2019-02-24:
Our unix builds call `python -m pip install python-can`
If you try that on a Unix machine right now, It will fail as the above (having python-can uninstalled first)

gpotter2, 2019-02-24:
Update:
- the issue ONLY happens when installing through wheels
- `git bisect` states that https://github.com/hardbyte/python-can/commit/eb8ee51d42399aca913401bc7d073b6e63aa5aa9 is causing the issue. I could replicate: it works before this commit

hardbyte, 2019-02-24:
Replicated on linux:

```
(test3-1-0) [brian@hardbyte-nzxt tmp]$ python -m pip install -U setuptools
Collecting setuptools
  Using cached https://files.pythonhosted.org/packages/d1/6a/4b2fcefd2ea0868810e92d519dacac1ddc64a2e53ba9e3422c3b62b378a6/setuptools-40.8.0-py2.py3-none-any.whl
Installing collected packages: setuptools
  Found existing installation: setuptools 40.6.2
    Uninstalling setuptools-40.6.2:
      Successfully uninstalled setuptools-40.6.2
Successfully installed setuptools-40.8.0
(test3-1-0) [brian@hardbyte-nzxt tmp]$ pip install python-can==3.1.0
Collecting python-can==3.1.0
  Using cached https://files.pythonhosted.org/packages/02/01/133f113b9914b26891f0c13be61823872f9216ab0fa1859f6f1b7a15c122/python_can-3.1.0-py2.py3-none-any.whl
Collecting typing (from python-can==3.1.0)
  Using cached https://files.pythonhosted.org/packages/4a/bd/eee1157fc2d8514970b345d69cb9975dcd1e42cd7e61146ed841f6e68309/typing-3.6.6-py3-none-any.whl
Collecting windows-curses (from python-can==3.1.0)
  Could not find a version that satisfies the requirement windows-curses (from python-can==3.1.0) (from versions: )
No matching distribution found for windows-curses (from python-can==3.1.0)
```

I found [a relevant comment](https://stackoverflow.com/questions/48048745/setup-py-require-a-recent-version-of-setuptools-before-trying-to-install#comment83091091_48049510) on stackoverflow but I'm still not sure what needs to change to support wheels.

hardbyte, 2019-02-24:
So I built the release last night using Python 3.5 and the `wheel` package v0.29.0 wasn't up to date. I've just rebuilt with Python 3.7 with the latest setuptools, pip and wheel packages. 

Available on pypi: `pip install -U python-can==3.1.1rc1`

gpotter2, 2019-02-24:
@hardbyte Thanks for the update !
I can confirm it fixes the issue. It would be great to have a release though, so that we don't have to update the versions manually.

gpotter2, 2019-02-25:
3.1.1 was released . Fixes the issue

Thanks for your quick support

https://pypi.org/project/python-can/3.1.1/

closed 2019-02-25 by gpotter2

mpena2099, 2020-10-10:
Hi! I think this bug is "active" again: 

_$ pip3 install python-can_
```
Defaulting to user installation because normal site-packages is not writeable
Processing /home/inforcomp/.cache/pip/wheels/23/22/aa/ab09f2d1a99925c9eb5c38e0afbb717c48fe59b74c4480d4ce/python_can-3.3.4-py2.py3-none-any.whl
Requirement already satisfied: typing in /home/inforcomp/.local/lib/python3.7/site-packages (from python-can) (3.7.4.3)
Requirement already satisfied: aenum in /home/inforcomp/.local/lib/python3.7/site-packages (from python-can) (2.2.4)
Requirement already satisfied: wrapt~=1.10 in /home/inforcomp/.local/lib/python3.7/site-packages (from python-can) (1.12.1)
ERROR: Could not find a version that satisfies the requirement windows-curses (from python-can) (from versions: none)
ERROR: No matching distribution found for windows-curses (from python-can)

```

_$ pip3 install python-can==3.1.1_
```
Defaulting to user installation because normal site-packages is not writeable
Collecting python-can==3.1.1
  Using cached python_can-3.1.1-py2.py3-none-any.whl (149 kB)
Requirement already satisfied: wrapt~=1.10 in /home/inforcomp/.local/lib/python3.7/site-packages (from python-can==3.1.1) (1.12.1)
Installing collected packages: python-can
Successfully installed python-can-3.1.1

```

meyerrap, 2021-06-02:
Can confirm last comment.

Version 3.3.3 seems to install just fine, still. 3.3.4 is broken again

felixdivo, 2021-06-02:
I'd propose just failing forward to the next release, 4.0.0 is coming up and a dev release it already available.