hardbyte/python-can #521: Release files shouldn't be manually created

hardbyte, 2019-02-24:
Ideally we configure our CI/CD setup to create the latest wheels and source dists and upload to pypi.

felixdivo, 2019-02-24:
I found this: https://docs.travis-ci.com/user/deployment/pypi/

hardbyte, 2019-02-26:
Ok I'm keen to give Travis some more responsibility :-)

Perhaps it can do the github releases at the same time - https://docs.travis-ci.com/user/deployment/releases/

In any case the first step is to:
- [ ] get travis to build the artifacts.
- [ ] test that the artifacts can be installed
- [ ] then we can configure travis to publish new releases (from the master branch)

acolomb, 2019-02-26:
Just a heads up, I haven't researched myself, but maybe it is not so wise to invest in Travis CI anymore:

https://blog.steve.fi/experimenting_with_github_actions.html

hardbyte, 2019-02-26:
Yeah I saw they were laying off staff after being acquired :-/ 

I think the discussion of moving off Travis-CI is independent of fixing the issue with me (or some other person) manually building the release artifacts. We could easily move to Circle-CI or github actions in the future, but for right now I'm happy to stick with travis.

felixdivo, 2019-02-26:
Moving from Travis to Docker would also be some additional effort if we want to keep socketcan working in the test environment. But I haven't further looked at it. Also, switching to Travis' new xenial environment poses the same problem, socketcan seems to be unavailable there. At least there were problems popping up on [that branch](https://travis-ci.org/hardbyte/python-can/jobs/488566990#L175).

felixdivo, 2019-04-06:
Now,

> test that the artifacts can be installed

still remains open. Do we really need it?

hardbyte, 2019-05-12:
Just recording this working for the first time for release 3.2.

First upload of tag `3.2.0a0` failed - presumable because of a manual or early test upload. 
```
Uploading distributions to https://upload.pypi.org/legacy/
Uploading python-can-3.2.0a0.tar.gz
100%|█████████████████████████████████████████| 183k/183k [00:00<00:00, 190kB/s]
NOTE: Try --verbose to see response content.
HTTPError: 400 Client Error: File already exists. See https://pypi.org/help/#file-name-reuse for url: https://upload.pypi.org/legacy/
```

Retrying now with `3.2.0a1`

hardbyte, 2019-05-12:
Ahh no I see the mistake, each build job (python2.7, python3.6 etc is trying to deploy... Need to adjust the travis file.

hardbyte, 2019-05-12:
Getting there with Travis stages:

[image]

hardbyte, 2019-05-13:
Seems to be working as advertised

PR #577 (2019-05-12): Use travis stages
This fixes #521 by switching travis-ci to use [stages](https://docs.travis-ci.com/user/build-stages/) instead of a matrix build.

In build https://travis-ci.org/hardbyte/python-can/jobs/531305592 travis deployed to PyPi https://pypi.org/project/python-can/3.2.0a2/ (I'd tagged the commit by creating a GitHub pre-release)

Also:
* added another target `pypy3`.
* added a pip cache.
closed 2019-05-13 by hardbyte