hardbyte/python-can #522: readthedocs.io "stable" links to version 2.2.1

felixdivo, 2019-02-25:
This: https://python-can.readthedocs.io/en/stable/ links to to documentation of python-can version 2.1.1 instead of 3.1.1. This problem exists for more than the most current release.

hardbyte, 2019-02-25:
Do you have a readthedocs account @felixdivo ? I can disable the `stable` site but can't seem to change what commit it points to. Perhaps it is done with git tags.

felixdivo, 2019-02-25:
No, I don't have an account there, so you probably know things better than me. But I found this: https://docs.readthedocs.io/en/stable/versions.html

hardbyte, 2019-02-26:
Got it thanks. Seems like readthedocs got stuck pointing at an old tag.

Anyhow under the release 3.1.1 I added the read the docs tag "stable" and that seemed to move it. Will have to see if it automatically updates to newer releases.

closed 2019-02-26 by hardbyte