hardbyte/python-can #529: How to use with Advantech CANbus on Windows?

mindfuucker, 2019-03-05:
Hello,

I bought an Advantech industrial PC with [dual isolated CANbus](https://www.advantech.com/products/14263729-aaa3-4552-b990-99d16cdfee24/pcm-26d2ca/mod_9a1e9dbf-e22d-4770-a896-cecf40607084) transceivers. It came pre-installed with Advantech branded drivers, the same drivers seem to be appear in these devices from Advantech: [link](http://advdownload.advantech.com/productfile/Downloadfile3/1-ICS73W/PCI-1680U_1682U_Manual_Ed1.pdf). 

It has support for a lot of programming languages including LabVIEW / C# / VB.NET and VC.NET. What I try to figure out is: 
1. What interface it is using?
2. Is this interface already supported by python-can?

felixdivo, 2019-04-22:
(I think nobody has a clue)

karlding, 2019-05-17:
On the website you linked, there seems to be a [SocketCAN driver](https://support.advantech.com/support/DownloadSRDetail_New.aspx?SR_ID=1-T8UXK9&Doc_Source=Download) under the "Documents & Downloads", which should be usable on Linux. Is your PCIe device supported by that driver?

If so, you can simply use the SocketCAN interface that python-can provides.

closed 2019-09-05 by felixdivo