hardbyte/python-can #530: Interprocess virtual interface

felixdivo, 2019-03-09:
The current `virtual` interface is of limited use since it only distributes messages within the same process. If we could make it work with multiple processes it could be used in cases where a system of multiple processes needs to be run or tested.

It would be simple to implement if we had a central daemon that did this, but avoiding this would be elegant since a user of the library would need to set up less and there would be no central entity that could have problems.

I imagined it to be like this: Process A sends `("my_network_1", can.Message(...))` to some multicast/broadcast system and Process B does `net, msg = broadcast.read(); if net == "my_network_1": process(msg)`.

Does someone have an ideas on how to implement such a multicast/broadcast, possibly in a cross-platform way? [Over here](https://wiki.python.org/moin/UdpCommunication#Multicasting.3F) in the Official Python Wiki, people said it is tricky to get right. Is that still the case? Should be use [zeroconf](https://github.com/jstasiak/python-zeroconf) instead (probably not)?

felixdivo, 2019-03-10:
One reason why we might want this feature: Testing software that interchanges messages within docker containers is currently not easily possible since Docker does not support socketcan-like communication within a single container, AFAIK. [This](https://medium.com/@tantony/using-virtualcan-inside-docker-on-macos-cfa10f7e75d6) seemed rather complicated + is not isolated. And the virtual socketcan interface (vcan) is not available on Windows/macOS as well.

felixdivo, 2019-03-17:
This demo from the official Python repo seems to work nicely: https://svn.python.org/projects/python/trunk/Demo/sockets/mcast.py

felixdivo, 2019-03-23:
@hardbyte @christiansandberg How should that interface be called, if I implement it?

1. I guess it should not replace the current `virtual` interface, since that one is still nice to have and work quite reliably I think.
2. How should we call it? `virtual-interprocess`?

windelbouwman, 2019-04-02:
I have the another use case. I want to test embedded code on a PC build, but still use CAN communication. My current approach is to create a zeromq server, which allows clients to connect. CAN messages are serialized by using msgpack. The zeromq server acts as a sort of virtual bus to which clients can connect.

felixdivo, 2019-05-20:
I started some draft on branch `add-interprocess-virtual-bus` but currently do not have time to complete it.

closed 2020-12-13 by hardbyte