hardbyte/python-can #542: When I try to set the CAN bus with CANoE Highspeed channel, the test failed with error:ValueError: negative shift count

cffznz, 2019-04-04:
I use the Python-can library integrate as robotframework keyword.
And when I set CAN bus in my test suite, after run the test suite about more than 1 min
The issue always occred as following:

12:45:05.426 | DEBUG | loaded can config: {'interface': u'vector', 'bitrate': 500000, 'app_name': u'CANoe', 'channel': u'0'} |  
12:45:05.442 | DEBUG | Channel index 0 found
12:45:05.465 | FAIL | ValueError: negative shift count
12:45:05.465 | DEBUG | Traceback (most recent call last):   File "C:\RF_RMS\CANMsg.py", line 15, in Set_Bus     self.bus=can.interface.Bus(bustype=Bustype, app_name=App, channel=ChannelNum, bitrate= Bitrate)   File "c:\python27\lib\site-packages\can\interface.py", line 117, in __new__     return cls(channel, *args, **config)   File "c:\python27\lib\site-packages\can\interfaces\vector\canlib.py", line 104, in __init__     mask = 1 << idx

The strange thing is the test with the same configuration of CAN in the before test step all are OK. After I run the test suit about after more than 1 min, this issue always occurred.
Seems the mask = 1 << idx had got some not correct value.
Could you help and check what happed in this? and help me to resolve this issue.
Thank you.

christiansandberg, 2019-04-04:
Which version are you using? Sounds like #353.

cffznz, 2019-04-08:
I use Vector CANoe 9.0 and High Speech 2 channel CAN Case.
And for Python can, it should be the latest version.
And Python, I use Python 2.7.16.

felixdivo, 2019-09-12:
Is it still present on `develop` (which is now using Python 3 and received quite a few bugfixes/changes)?

felixdivo, 2021-04-14:
This is stale, old and no reproducible error is provided. Please open a new issue if it persists.

closed 2021-04-14 by felixdivo