hardbyte/python-can #543: Why use socketcan.constants rather than those provided by socket?

bessman, 2019-04-04:
To provide some background, I'm working with a slightly non-standard SocketCAN implementation in an embedded system. Among other things, it implements its own CAN_RAW and CAN_BCM protocols, does not support error frames or CAN FD. plus some other idiosyncracies. As such, I've had to make a few minor changes to can.interfaces.socketcan to get it working.

One of the changes I had to make was to [line 465 in socketcan.py](https://github.com/hardbyte/python-can/blob/develop/can/interfaces/socketcan/socketcan.py#L465) to use socket.SOL_CAN_RAW (which evaluates to 101) instead of SOL_CAN_RAW from constants.py (which evaluates to 107).

I'm curious what the reason is to use the constants from constants.py rather than those provided by socket?

christiansandberg, 2019-04-04:
Those constants are not available in the socket module in Python 2.7. Although I don’t see how it could become 107... 🤔

bessman, 2019-04-04:
Ah, that's because that was one of the other things I changed. In my system, CAN_RAW is 7 instead of 1. I see now that SOL_CAN_RAW = SOL_CAN_BASE + CAN_RAW, so that's why it's 107.

closed 2019-04-04 by bessman