hardbyte/python-can #544: Failing "setup.py test" on develop branch after 3.1.1 in Windows

bhass1, 2019-04-04:
OS: Windows 7
Python: 3.6.1
Branch: develop
Commit: dd89de4974d.....

Description:
Windows doesn't like the changes between release-3.1.1 and referenced branch+commit above. 

Running "py -3 setup.py test" from the tip of branch "release-3.1.1" completes successfully. The same exact command run on the above referenced branch+commit fails with the following. See image below.

[image]

I haven't tried to root-cause the issue yet. Appears one of the newly supported interfaces causes a hiccup.

bhass1, 2019-04-04:
I need to downgrade pytest from 4.3.0 to 3.6.0 before "setup.py test" will work from release-3.1.1 branch. But after downgrading pytest and using  the release-3.1.1 branch on the same machine, the same area of output looks like the following:
[image]

bhass1, 2019-04-04:
The two differences are:
WARNING:can.interfaces.iscan:Failed to load IS-CAN driver: [WinError 126] The specified module could not be found
ERROR:can.interfaces.nican:Failed to load NI-CAN driver: [WinError 126] The specified module could not be found

felixdivo, 2019-04-07:
Travis CI and APpveyor call our tests using the command `pytest`. Does that work on your machine?

felixdivo, 2019-04-07:
Could it be, that your test runner executes the generic `ReaderWriterTest` as a test, and ignoring the `__test__ = False` attribute? Because all other tests that override that class also seem to implement `@abstractmethod def _setup_instance(self)`.

bhass1, 2019-04-08:
Using pytest version 4.3.0 from "pip install pytest==4.3.0" : 
[image]

felixdivo, 2019-04-09:
Looks like you did not install all dependencies there. Please install them with:

```bash
pip install .[test]
```

[These are missing](https://github.com/hardbyte/python-can/blob/develop/setup.py#L31) and causing the problems.

felixdivo, 2019-04-09:
here is the entire diff: https://github.com/hardbyte/python-can/compare/a6a4447ed9477afa99bb366e83a1c12d330789fb...dd89de4974d141a0b46c179133679b667669e185

bhass1, 2019-04-09:
`pip install .[test]` did the trick for running `pytest` to completion without error.

`py -3 setup.py test` is still failing at the same spot.

Perhaps the `setup.py test` should be deprecated for the use of `pytest` instead? In either case, I think it would be helpful if the documentation was updated to include `pip install .[test]`.

Regarding the `setup.py test` issue... what makes you think it is involved with `ReaderWriterTest`? Is there something I can try to help diagnose the root cause of the error?

felixdivo, 2019-04-10:
> I think it would be helpful if the documentation was updated to include pip install .[test].

This is actually explained [here](https://python-can.readthedocs.io/en/stable/development.html#building-installing):
"_Maybe, you need to execute pip3 install python-can[test] (or only pip for Python 2), if some dependencies are missing._"

felixdivo, 2019-04-10:
I have looked in the [docs of pytest](https://docs.pytest.org/en/latest/goodpractices.html#integrating-with-setuptools-python-setup-py-test-pytest-runner), and found that this might be missing:

```
[aliases]
test=pytest
```

Does that solve the errors?

felixdivo, 2019-04-10:
> what makes you think it is involved with ReaderWriterTest?

Hm, all tests in that file inherit from that class, which is itself not executed as a test, because it has that `__test__ = False` set. And the `NotImplementedError` would only be raised if 
(a) that test class is executed directly (not only serving as a superclass), or
(b) some other tests is extending it but does not override `_setup_instance()`, or
(c) for some reason not the overridden method but the original one is called (this would be the strangest I think ^^).

bhass1, 2019-04-10:
> This is actually explained here:
"Maybe, you need to execute pip3 install python-can[test] (or only pip for Python 2), if some dependencies are missing."

Does `pip3 install python-can[test]` do the same thing as `pip3 install .[test]`? I thought I ran the former before complaining about pytest not working, but I can't remember. I definitely saw the doc you are pointing at. What about adding some clarity on what errors look like when dependencies are missing? My error wasn't very clear to me that it meant missing dependencies...

>I have looked in the docs of pytest, and found that this might be missing:
[aliases]
test=pytest
Does that solve the errors?

It got me close! I also had to add `setup_requires=["pytest-runner"]` in `setup.py`. With those changes, calling `py -3 setup.py test` runs pytest to completion without error.

But that only covers up the underlying issue with the unittest error. I tried digging deeper by adding print statements. Sorry if it's not helpful, but the best I got was this additional info: 
```
@abstractmethod
def _setup_instance(self):
    print(str(self)) #  < ------ added print -----------
    raise NotImplementedError()
``` 
outputs: `test_append_mode (test.logformats_test.ReaderWriterTest)`

felixdivo, 2019-04-11:
> Does pip3 install python-can[test] do the same thing as pip3 install .[test]?

They are similar: The first one downloads it from PyPi and second one installs it from a local folder. But both install the "test" extra.

> What about adding some clarity on what errors look like when dependencies are missing?

Well, that's difficult I think. That error above is quite bizarre ...

> It got me close! I also had to add setup_requires=["pytest-runner"] in setup.py. With those changes, calling py -3 setup.py test runs pytest to completion without error.

That's great. We should add that to the library.

> But that only covers up the underlying issue

Yes, we should look further into that as well! It looks like `test.logformats_test.ReaderWriterTest` is run as a test despite being marked as `__test__ = False`.

felixdivo, 2019-04-14:
https://docs.pytest.org/en/latest/nose.html#supported-nose-idioms that attribute should work ... Is there a better way to avoid this parameter all together?

felixdivo, 2019-04-15:
Do you have time for submitting a PR with the fix for calling `python setup.py test`?

felixdivo, 2019-04-27:
[Here](https://stackoverflow.com/questions/1323455/python-unit-test-with-base-and-sub-class) is a rather long StackOverflow post on the problem of providing a common base class for `TestCase`s that is not executed itself. I'll have a look it it.

PR #558 (2019-05-03): Fix broken execution of tests on some Windows environments
This fixes the non-standard usage of `__test__ = False` to exclude the base class from execution and uses a simple `del` statement instead. The `__test__` attribute was a Nose idiom (though supposedly supported by pytest 🤷‍♂️). But as we don't use Nose any more, this seems to be the better way!

Fixes #544. (@bhass1 could you verify this, since you can reproduce the error nicely?)
closed 2019-05-03 by hardbyte