hardbyte/python-can #549: How to change SocketCan bitrate at runtime?

mauriciocirelli, 2019-04-11:
Dear,

I would like to implement an auto-baud feature in my application. The idea is to probe the bus for incoming messages until we find the right bitrate from a list.

Is it possible to do that using socketcan interface? If so, how could I do that? Do I have to create new bus instances everytime I change the bitrate or can I reuse the same instance?

Thank you!

felixdivo, 2019-04-18:
This is how you set the bitrate: https://python-can.readthedocs.io/en/stable/interfaces/socketcan.html#real-device. I don't think that can be done with syscalls when that interface has already been created but I didn't look at the socketcan specification.

mauriciocirelli, 2019-04-18:
I see.

I was hoping for a Python API. I think I will have to run this shell
command from my Python app.

Thank you.

hardbyte, 2019-04-20:
Maybe we could make an abstract driver interface and implement socketcan calls as a Python api.

mauriciocirelli, 2019-04-22:
It seems that it is possible using ioctls:

https://stackoverflow.com/questions/45967947/setting-socketcan-bitrate-programmatically

The source code for libsocketcan is here:
https://github.com/lalten/libsocketcan/blob/gh-pages/src/libsocketcan.c

There is a can_set_bitrate function that use system calls to setup the netlink.

felixdivo, 2019-05-02:
> I would like to implement an auto-baud feature in my application

You could also implement this in the library

iurly, 2019-10-13:
Hi @mauriciocirelli, 

I'm also interested in such a feature, did you manage to get around this issue?
So far I've been calling `sudo ip link` (which would be the equivalent of what libsocketcan does) using `subprocess.call()` , but that raises a few issues with `sudo` being required and configured.
I assume the same would apply to the equivalent syscalls made from within the python application (whether from pure-Python or a C extension).
Did anyone get around this somehow?
Thanks!

mauriciocirelli, 2019-10-13:
Hi,

I am doing the same... The difference is that I do not need (nor have) sudo
as I am working on a embedded system.
But it would be great to have such feature in the API.

Mauricio Cirelli
www.futuragames.com.br

cperkulator, 2021-02-17:
> Maybe we could make an abstract driver interface and implement socketcan calls as a Python api.

So would a PR be welcomed to do something like:
```
def change_baudrate(self, baudrate):
    raise NotImplementedError()
```
in the BusABC.

Also when it relates to changing the baudrate on socketcan, it looks like are a couple options for changing the baudrate:

- [libsocketcan](https://github.com/lalten/libsocketcan/blob/gh-pages/src/libsocketcan.c)
- [pysocketcan](https://github.com/Tbruno25/pysocketcan/blob/master/pysocketcan/pysocketcan.py)
- [netlink sockets directly](https://stackoverflow.com/questions/45967947/setting-socketcan-bitrate-programmatically)

Does anyone have any experiences with these?  Or is there a suggestion for a place to start with switching baudrate on socketcan?