hardbyte/python-can #552: Canal is 32bit, but 32bit USB2CAN driver won't install on my Windows 10

dposea, 2019-04-17:
My USB2CAN device will not work with the 32 bit driver. I followed the 8devices instructions exactly, and tried a few other ways. The 64bit driver installed the first time. I have 64bit python installed already, so a 84bit usb2can.dll would really be helpful. I could only find one repository for CANAL. I tried to build a 64 bit canal dll from krumboecks GitHub , but it has no project files and I could not get a good compile trying to choose options. I haven't tried this in a 32bit windows VM yet. I  had problems using USB2CAN In a Linix VM, and I need to sniff some windows reflash programs, I have extensive knowledge of Ford and GM reflashing and tuning, and I'd like to be helpful if possible, if I can ever git CAN sniffing to work on Windows.

acolomb, 2019-04-17:
One of the original developers of the USB2CAN device kindly provided a 64 bit version of usb2can.dll in #465.

He offered it to 8devices and I contacted them several times asking for it to be included on the download page (also for their new Korlan devices), but never heard back. Seems like they are not interested in modern computing 😉. You could try mailing them to increase the pressure...

acolomb, 2019-04-17:
Note about sniffing: I tried their instructions to integrate an extcap helper in Wireshark for the USB2CAN recently. That didn't work however, it seems to not detect any adapter, regardless of which usb2can.dll version I tried.

Python-can works alright with it though, once you get the 32 / 64 bit stuff sorted out.

rusoku, 2019-04-17:
Hi,
can you provide kernel driver error code ?
Really usb2can doesn't have kernel driver for Windows 10.Last kernel driver was developed by me only for Windows 7 and Windows XP only.And 8devices are resellers only for usb2can.
Canal dll for both 64 and 32 version are there: https://www.rusoku.com/downloads/usb2can_dll_1.0.0.zip
Recently I launched my own brand. Brand name is Rusoku (www.rusoku.com)
Converter kernel drivers are based on WinUSB generic driver.
SocketCAN available also.
And at the moment I am working on CAN FD device

acolomb, 2019-04-17:
@rusoku From my experience, the kernel driver is not a problem actually. As the OP specifically asked for a 64-bit version of usb2can.dll, I think it is the same problem I ran into. When running a 64-bit Python executable, the pywin32 module will fail to dynamically link to a 32-bit usb2can.dll.  This is noted in the [backend documentation for USB2CAN](https://python-can.readthedocs.io/en/stable/interfaces/usb2can.html#windows-install).

Refraining to a 32-bit Python installation is not a workable solution nowadays...

dposea, 2019-04-18:
I finally got rusoku's code to compile, then the linker failed. While looking for a solution to that I found that the vscp project has a newer version of the Canal usb2can interface on GitHub. It's buried under many directories, Here is a link : 
https://github.com/grodansparadis/vscp/tree/master/src/vscp/drivers/level1/usb2can/usb2can

I was able to build that version in 64 bit mode. Thanks to all  who replied, especially acolomb and rusoku .
Oh yeah, under that directory you'll find a complete Visual Studio project, so building is easy. I may have a go at adding filters for usb2can. Are there other devices I should consider getting that are not too expensive. Something with good windows support because everything  I need to sniff runs on windows and I don't want to use two laptops. 

Just to be very clear, my problem was the 32 bit driver would not install sucessfully at all on 64 bit windows, I may have been able to run 32bit python and 32 bit Canal, but the driver was going to be 64 bit always. Seriously, any software still being devloped should support 64bit code. I realize no one here actually caused the problem, 8devices did by dropping the ball entirely. Again thanks for the help, and any advice on a better or better supported device.

rusoku, 2019-04-18:
ok.Fine.
This is not latest source code but should work.Sometimes can freeze because treads are terminated incorrectly, doesn't suport usb hot-unplug, etc.
Becauce vscp canal api standard doesn't support can frames filtering for TouCAN converters line I added extended functions CanalSetFilter11bit and CanalSetFilter29bit .Now hardware filtering works fine.
https://www.rusoku.com/storage/app/media/CANAL%20Library%20User%20Guide.pdf

Gediminas

dposea, 2019-04-19:
Where can I get a TouCAN? Probably more than one for my company. USB2CAN is not very widely used by other CAN open source tools like Busmaster, Kasvr, etc. Even worse I bought a Canalyze  that only uses the 8devices driver. So far it has worked as a USB2CAN device except win64.

rusoku, 2019-04-19:
USB2CAN was developed for my VSCP home system.At the beginning it was not intended for car diagnostic.Because microcontroller STR750 now isn't longer produced I redesigned new USB to CAN converter (TouCAN) based on STM32F412.It has much more larger buffers for CAN frames.
TouCAN converter you can at get https://www.rusoku.com/shop/
or from Ebay.
Or if you working on interesting open source project I can sent one device for free ;)

Gediminas

povk, 2019-04-24:
You can get the 64 bit CANAL library from 8devices now:
https://www.8devices.com/products/usb2can_korlan#downloads

felixdivo, 2019-05-12:
Can this be closed or are there changes required in this library, like documentation?

rusoku, 2019-05-12:
imho can be closed

closed 2019-05-12 by felixdivo

dposea, 2019-05-14:
I'd like to know for sure if a Toucan will work on 64bit windows 10, and when they will be back in stock. If the company owner still helps out customers of another company it's an easy decision which product to buy next. USB2CAN just doesnt work except on linux, and it now discontinued. I never got it working under windows, but I also found out my hardware was a clone running different firmware. It is called a Canalyze, uses 8devs drivers, but has it own firmware and I could never get it to update with dfu or defuse