hardbyte/python-can #557: how to create a script candump and cansend

maltsevvv, 2019-04-19:
how to create a script python can0 candump and cansend with execution if.
example:
if candump 448 # B1 then cansend 264 # B1

hardbyte, 2019-04-20:
What have you got so far?

maltsevvv, 2019-04-20:
Hey.
I have Audi Navigation (RNS-D) + tv-nuner + raspberry pi + mcp2515.
When it's all hooked up, I control the raspberry, the buttons on the RNS-D by can.
All perfectly.

Now I want to remove the tv-tuner and emulate the TV tuner for
Audi Navigation (RNS-D),
so it should be
Audi Navigation (RNS-D) + raspberry pi + mcp2515.

create `sudo nano script /home/pi/tvtuner.py`
` #!/home/pi/tvtuner.py
import os
import time
import can

DELAY_PERIOD = 0.3 #delay in sending
timeLastSend = time.time()
isRunning = True

while(isRunning):

    if(time.time() - timeLastSend >= DELAY_PERIOD):
        os.system("cansend can0 264#A101") #tv activation
    if(time.time() - timeLastSend >= DELAY_PERIOD):
        os.system("cansend can0 264#11010100000000") #control activation
    if(time.time() - timeLastSend >= DELAY_PERIOD):
        os.system("cansend can0 264#10150100020000") #control activation
    if(time.time() - timeLastSend >= DELAY_PERIOD):
        os.system("cansend can0 264#B0")
    if(time.time() - timeLastSend >= DELAY_PERIOD):
        os.system("cansend can0 264#B1")
    if(time.time() - timeLastSend >= DELAY_PERIOD):
        os.system("cansend can0 264#B2")
    if(time.time() - timeLastSend >= DELAY_PERIOD):
        os.system("cansend can0 264#B3")
    if(time.time() - timeLastSend >= DELAY_PERIOD):
        os.system("cansend can0 264#B4")
    if(time.time() - timeLastSend >= DELAY_PERIOD):
        os.system("cansend can0 264#B5")
    if(time.time() - timeLastSend >= DELAY_PERIOD):
        os.system("cansend can0 264#B6")
    if(time.time() - timeLastSend >= DELAY_PERIOD):		
        os.system("cansend can0 264#B7")
    if(time.time() - timeLastSend >= DELAY_PERIOD):
        os.system("cansend can0 264#B8")
    if(time.time() - timeLastSend >= DELAY_PERIOD):
        os.system("cansend can0 264#B9")
    if(time.time() - timeLastSend >= DELAY_PERIOD):
        os.system("cansend can0 264#BA")
    if(time.time() - timeLastSend >= DELAY_PERIOD):
        os.system("cansend can0 264#BB")
    if(time.time() - timeLastSend >= DELAY_PERIOD):
        os.system("cansend can0 264#BC")
    if(time.time() - timeLastSend >= DELAY_PERIOD):		
        os.system("cansend can0 264#BD")
    if(time.time() - timeLastSend >= DELAY_PERIOD):
        os.system("cansend can0 264#BE")
    if(time.time() - timeLastSend >= DELAY_PERIOD):
        os.system("cansend can0 264#BF") `

`python tvtuner.py`

everything is working
but the CPU loads 100%
Without this script cpu 20%
 
This video of the TV tuner
[https://github.com/maltsevvv/raspberry/raw/master/carpc/rns-d/rns-d%2Btv-tuner%20can.mp4](url)

maltsevvv, 2019-04-20:
button id [https://raw.githubusercontent.com/maltsevvv/raspberry/master/carpc/rns-d/can_id.txt](url)

hardbyte, 2019-04-21:
So from what you have shared you're not currently using the Python library `can`, just calling the `cansend` command from socketcan tools. You could follow our documentation and write a Python script to efficiently carry out what you're doing using the send periodic feature e.g.

```python
msg = can.Message(arbitration_id=0x264, data=[0xb8], extended_id=False)
task = bus.send_periodic(msg, 0.30)
```

Or you could fix up your system call script to be a bit more efficient. Instead of a loop that checks millions of times per second if `DELAY_PERIOD` has elapsed you probably want to sleep for `DELAY_PERIOD`, and then you don't have to check the `time()` multiple times in the loop.

I'm going to close this issue because it isn't an issue with this library, but if you do end up using `can` and have issues please reopen or create a new issue. Best of luck!

closed 2019-04-21 by hardbyte