hardbyte/python-can #560: Handling of errors of ixxat can interface

marcel-kanter, 2019-04-29:
Hi,
for the ixxat backend, all the errors, that may occur are just logged, including stuffing, ack and crc errors. These should be rare but can occur and should be handled.

This is an TODO for the ixxat backend in the _recv_internal method.

Are there plans to update the implementation?
Should I try to fix this (this may involve a change in the return of the _recv_internal and adding some general exceptions to the can interface)?
I would change it to something like this, the method should return a message, None or raise an exception.

wait until timeout is over:
  if message received:
    return message
  if ack, stuff, etc error:
    raise exception
  if timeout (in the interface):
    pass

christiansandberg, 2019-04-29:
As it works for most interfaces today is that fatal errors that causes the interface to shut down and not receive any more messages may raise exceptions but “normal” bus issues that may occur like those you mention are represented as a Message with is_error_frame=True. The problem is that there is currently no interface agnostic way to know what type of error frame it is.

marcel-kanter, 2019-04-29:
Take the ACK error for example. It is caused when only one bus member is at the bus and sends an message. Since it does not ack its own message the ack error occur. The error frame has nothing to do with a missing ack.
Error frames are send out, when framing errors where detected on the bus. This is not the case with only one bus member.
Maybe the other bus members have been disconnected by an cable break.
BTW. I've started to address this in conjunction with the unification of the exceptions.

marcel-kanter, 2019-04-30:
Ok. The is_error_frame=True solution works for the receiver side. I'll try to update the ixxat interface accordingly.
But how should a send behave?
Take the minimal example:
```
import can
import time
if __name__ == '__main__':
	theBus = can.Bus(interface = 'ixxat', channel = 0, bitrate = 250000)
	for i in range(10):
		print("send message")
		theBus.send(can.Message())
		time.sleep(1)
```
This code runs fine, but no exception or error is reported. I would expect an exception or some kind of hint (an exception).

Addtionally this issue is related to issue #356

christiansandberg, 2019-04-30:
You can provide a timeout as the second argument to send. If the message has not been acked within this time an exception is raised. The behavior varies between interfaces though, some raise error when the buffer is full and some do not support it at all.

marcel-kanter, 2019-04-30:
Ok. I tried this already. But wouldn't it make sense, if the error is raised, if the interface is in an error state too, without an timeout?

christiansandberg, 2019-04-30:
Just because some messages are not being acked doesn’t mean the interface is in error. It might be normal, like if you’re the first node to wake up after a system starts or in a lab environment where you connect and disconnect equipment from time to time.

Most applications I think just displays the ACK error frame as a warning but continues to function.

felixdivo, 2019-06-12:
@marcel-kanter Can this be closed?

marcel-kanter, 2019-06-12:
Yes. Can be closed.

closed 2019-06-13 by felixdivo

felixdivo, 2023-09-28:
@prunith This issue is closed. Please open a new one for a new issue.

PR #562 (2019-05-19): General exception handling (+ changes to the ixxat interface)
A starting point for the unification of the exception handling.
Includes some improvements for the ixxat interface (the improvements use the new exceptions).
Please pull this into new branch, to keep the develop working until this work is applied to all the interfaces/classes/etc...

Closes #560
Addresses #356