hardbyte/python-can #584: Test player.py

hardbyte, 2019-05-16:
See coverage at release 3.2 - https://codecov.io/gh/hardbyte/python-can/src/release-3.2.0/can/player.py

Ideally this is covered by a CI executed test on each supported version of Python.

Ralph-Ne, 2019-08-08:
Hi hardbyte,

I've got some problems with the newest version of the player in python-can.

The call is the same: python -m can.player -b 250000 -c can0 -v -i socketcan FILE_TO_PLAY

If I use version 2.2.1dev0, downloaded over a year ago, the player works fine, even without the -g option.

If I use the newest version 3.3.1, the player ends with:

Timestamp: 1504177560.972875        ID: 0000    S E DLC:  0                                Channel: 0
Timestamp: 1504177560.973315        ID: 0000    S E DLC:  0                                Channel: 0
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/usr/lib/python3.7/site-packages/can/player.py", line 98, in <module>
    main()
  File "/usr/lib/python3.7/site-packages/can/player.py", line 89, in main
    bus.send(m)
  File "/usr/lib/python3.7/site-packages/can/interfaces/socketcan/socketcan.py", line 633, in send
    raise can.CanError("Transmit buffer full")
can.CanError: Transmit buffer full
[canplayer@canplayer-pc ~]$

If I use the -g option it works, if I do -g 0.01: python -m can.player -b 250000 -c can0 -v -i socketcan -g 0.01 FILE_TO_PLAY,

-g 0.001 or -g 0.0001 lead to the same error as above.

Can you imagine what I've done wrong? Do you have any advice for me to solve the problem?

Kind regards

Ralph

karlding, 2019-08-09:
@Ralph-Ne can you open a new issue with the problem you're describing, and we'll triage from there? This issue is tracking some code coverage improvements.

Thanks!

felixdivo, 2021-04-23:
As discussed on the mailing list, we will mainly release what is there on release day. Only large issues are coordinated. Therefore, this issue is removed from the milestone.

pkess, 2022-01-16:
I think with #1198 this issue can be closed now. If you want additional tests just let me know

closed 2022-01-16 by zariiii9003