hardbyte/python-can #586: Improved testing of socketcan

hardbyte, 2019-05-16:
Testing coverage is approx 60% at release 3.2, shouldn't be too hard to get >90%.

A good start would be adding tests for BCM and ensuring the socketcan tests run on Python 2 (without native python support). A cheeky way to achieve better coverage would be dropping all the Python 2 code...

https://codecov.io/gh/hardbyte/python-can/src/7268f49f4ca349464e29bc680e9e42c66e313455/can/interfaces/socketcan/socketcan.py

felixdivo, 2019-05-16:
> A cheeky way to achieve better coverage would be dropping all the Python 2 code ...

That's what we will now to all over the library, if we work on any parts of it, right?

hardbyte, 2019-05-17:
Yup!

felixdivo, 2019-05-18:
Related: #590

felixdivo, 2019-07-15:
The main things [that are not tested that much](https://codecov.io/gh/hardbyte/python-can/src/develop/can/interfaces/socketcan/socketcan.py) are `MultiRateCyclicSendTask` and error/exception handling.

felixdivo, 2021-04-23:
Over time, there has been significant effort to improve testing (in socketcan). It's now at 87.84% statement coverage. This makes that interface one of the better tested ones. I don't really see the point in having an issue for tracking the testing of that specific interface at this stage. I suggest closing it.

closed 2021-07-14 by felixdivo