hardbyte/python-can #589: Cannot create or write file .blf

hoangkhai6893, 2019-05-17:
I try use can.BLFWrite('filelog.blf').on_message_received(msg) to create file , but it not working as well. the file is empty. I don't know how to fix it. Can you help me please ?

hardbyte, 2019-05-17:
Can you share your code? Try just using the can logger script `python -m can.logger -f filelog.blf`

christiansandberg, 2019-05-17:
You must also call the stop() method.

felixdivo, 2019-05-18:
> You must also call the stop() method.

Use use `can.BLFWriter` as a context manager.

hoangkhai6893, 2019-05-19:
Dear @hardbyte , this is my code . after i run that program, the CAN_log.blf is empty. I don't know why !

from __future__ import absolute_import

import can
bus = can.Bus(interface = 'socketcan',channel = 'can0', receive_own_message = True)
write = can.BLFWriter('/home/pi/Wiless_Logging_SYS/CAN_log.blf')
for msg in bus:
    write.on_message_received(msg)
    print(msg)

christiansandberg, 2019-05-19:
Here's a fourth answer for you:

```python
import can
bus = can.Bus(interface = 'socketcan',channel = 'can0', receive_own_message = True)
write = can.BLFWriter('/home/pi/Wiless_Logging_SYS/CAN_log.blf')
try:
    for msg in bus:
        write.on_message_received(msg)
        print(msg)
finally:
    write.stop()
```

christiansandberg, 2019-06-02:
Please re-open if you have any follow up questions.

closed 2019-06-02 by christiansandberg

itzranjeet, 2022-01-18:
```
write = BLFWriter(path + blf_file)
        print(msg)
        try:
            write.on_message_received(msg)
        finally:
            write.stop()
```
This writes to the same line.
I MEAN DATA IS REWRITTEN.

felixdivo, 2022-01-18:
Maybe you want to pass `append=True`? See [here](https://python-can.readthedocs.io/en/develop/listeners.html#can.BLFWriter). If not, please open a new issue. This one is 2.5 years old and closed.

AntonioMembrino, 2023-05-25:
> Maybe you want to pass `append=True`? See [here](https://python-can.readthedocs.io/en/develop/listeners.html#can.BLFWriter). If not, please open a new issue. This one is 2.5 years old and closed.

I am having problems with append=True.
 header = FILE_HEADER_STRUCT.unpack(data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
struct.error: unpack requires a buffer of 72 bytes

I have created a blf file in this way 
filenameblf = open("can_data"+str(new_dt)+".blf", "wb+")

felixdivo, 2023-05-26:
Again, this issue is very old! Please open a new one (possibly referencing this). It is best if you provide a reproducer script that one can run as-is. If you can attach the problematic file, even better.