hardbyte/python-can #591: Look into CI testing on Arm with shippable

hardbyte, 2019-05-17:
http://docs.shippable.com/platform/tutorial/workflow/run-ci-builds-on-arm/

> It is the only hosted CI/CD platform to offer native CI/CD for Arm developers.
> Even better news? Building on Arm machines is absolutely free for open source projects!

marcel-kanter, 2019-05-23:
sounds nice. At our company we use some fpgas with arm hard cores and linux runs on it.
I thought about a file in the doc showing the  platform support for each interface some weeks ago. Some interfaces support Windows only. That should be changed. I hope I have enough time to implement linux support (x86 and arm) for the ixxat interface for the 4.0 release. My not so secret plan is to use this interface as an example for the exception handling and multiplatform support.

hardbyte, 2019-05-25:
Many of the windows only interfaces can be used on Linux via socketcan. Documenting the support would be a good idea though