hardbyte/python-can #593: Broken Install - PyPI Release 3.2.0

phil-marble, 2019-05-17:
Related to #520 

Looks like a similar issue has com up in the latest release?

```$ chroot /tmp/make_image.26839.root /usr/local/bin/pip2 install python-can evdev
chroot: cannot change root directory to '/tmp/make_image.26839.root': Operation not permitted
phil@phil-870-120:~/code/mbot-build/images/mbotos$ sudo chroot /tmp/make_image.26839.root /usr/local/bin/pip2 install python-can evdev
Requirement already satisfied: python-can in /usr/local/lib/python2.7/dist-packages
Collecting evdev
  Using cached https://files.pythonhosted.org/packages/fe/6e/3fa84a43571dec4d00dc26befccc9867b6b3263651531cbc1345f718860f/evdev-1.2.0.tar.gz
Requirement already satisfied: typing in /usr/local/lib/python2.7/dist-packages (from python-can)
Collecting windows-curses (from python-can)
  Could not find a version that satisfies the requirement windows-curses (from python-can) (from versions: )
No matching distribution found for windows-curses (from python-can)
You are using pip version 9.0.1, however version 19.1.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

hardbyte, 2019-05-17:
Gah I don't like repeating mistakes. Thanks for reporting.

Assuming it is the same issue I've pushed a branch that updates `setuptools` and `wheel` before publishing to PyPi. Once [this](https://travis-ci.org/hardbyte/python-can/jobs/534035146) travis-ci job is complete there should be a new alpha on pypi.

Can you/anyone please confirm if that works and then I'll remove the alpha.

```
pip install -U python-can==3.2.1-alpha.1
```

phil-marble, 2019-05-17:
Actually, I can't seem to replicate the issue in a virtualenv:
```
(test001) phil@phil-870-120:~/tmp/test001$ pip install -U python-can==3.2.0
DEPRECATION: Python 2.7 will reach the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 won't be maintained after that date. A future version of pip will drop support for Python 2.7.
Collecting python-can==3.2.0
  Downloading https://files.pythonhosted.org/packages/7e/b2/84b5926a19a7a45602c14671defc8cbe0dbc55a8df122a12ff31e83632d0/python-can-3.2.0.tar.gz (174kB)
     |████████████████████████████████| 184kB 6.2MB/s 
Collecting wrapt~=1.10 (from python-can==3.2.0)
  Downloading https://files.pythonhosted.org/packages/67/b2/0f71ca90b0ade7fad27e3d20327c996c6252a2ffe88f50a95bba7434eda9/wrapt-1.11.1.tar.gz
Collecting aenum (from python-can==3.2.0)
  Downloading https://files.pythonhosted.org/packages/44/c4/7cda7e7e71e18ad999cb9d23f9969f818af1ed6be71d18db7963680b8320/aenum-2.1.2-py2-none-any.whl
Collecting typing (from python-can==3.2.0)
  Downloading https://files.pythonhosted.org/packages/cc/3e/29f92b7aeda5b078c86d14f550bf85cff809042e3429ace7af6193c3bc9f/typing-3.6.6-py2-none-any.whl
Building wheels for collected packages: python-can, wrapt
  Building wheel for python-can (setup.py) ... done
  Stored in directory: /home/phil/.cache/pip/wheels/b1/43/e4/932e100d8c53af23e4b1d67ed255dc5d7ade089b41678821ef
  Building wheel for wrapt (setup.py) ... done
  Stored in directory: /home/phil/.cache/pip/wheels/89/67/41/63cbf0f6ac0a6156588b9587be4db5565f8c6d8ccef98202fc
Successfully built python-can wrapt
Installing collected packages: wrapt, aenum, typing, python-can
Successfully installed aenum-2.1.2 python-can-3.2.0 typing-3.6.6 wrapt-1.11.1
```

phil-marble, 2019-05-17:
I'm attempting to reproduce in the original way I saw the issue...

phil-marble, 2019-05-20:
Repro:
```
$ sudo chroot /tmp/make_image.19320.root /usr/local/bin/pip2 install python-can==3.2.0
Requirement already satisfied: python-can==3.2.0 in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied: typing in /usr/local/lib/python2.7/dist-packages (from python-can==3.2.0)
Collecting windows-curses (from python-can==3.2.0)
  Could not find a version that satisfies the requirement windows-curses (from python-can==3.2.0) (from versions: )
No matching distribution found for windows-curses (from python-can==3.2.0)
You are using pip version 9.0.1, however version 19.1.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```

Verified Fix:
```
$ sudo chroot /tmp/make_image.19320.root /usr/local/bin/pip2 install python-can==3.2.1a0
Collecting python-can==3.2.1a0
  Cache entry deserialization failed, entry ignored
  Downloading https://files.pythonhosted.org/packages/97/60/0aa38d9d83acd14865eff6f0750ebd48255e5634f2f242e17b5fe7f67008/python-can-3.2.1a0.tar.gz (175kB)
    100% |████████████████████████████████| 184kB 2.4MB/s 
  Ignoring windows-curses: markers 'platform_system == "Windows"' don't match your environment
Requirement already satisfied: wrapt~=1.10 in /usr/local/lib/python2.7/dist-packages (from python-can==3.2.1a0)
Requirement already satisfied: aenum in /usr/local/lib/python2.7/dist-packages (from python-can==3.2.1a0)
Requirement already satisfied: typing; python_version < "3.5" in /usr/local/lib/python2.7/dist-packages (from python-can==3.2.1a0)
Building wheels for collected packages: python-can
  Running setup.py bdist_wheel for python-can ... done
  Stored in directory: /home/phil/.cache/pip/wheels/ed/37/50/ece4d761b960eb1a3b59b217ce0e869cafc5df4a547b117e6a
Successfully built python-can
Installing collected packages: python-can
  Found existing installation: python-can 3.2.0
    Uninstalling python-can-3.2.0:
      Successfully uninstalled python-can-3.2.0
Successfully installed python-can-3.2.1a0
You are using pip version 9.0.1, however version 19.1.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
```

phil-marble, 2019-05-22:
`Successfully installed python-can-3.2.1a0`

Your alpha release worksfor me.

mheden, 2019-05-24:
Both `3.2.1a0`  and `3.2.1.a2` seems to be broken for me with python3.6 on linux (`windows-curses` issue)

far3ll, 2019-06-04:
I'm on Windows 10 with Python 3.7 and I'm also stuck in the upgrade process (same for both alpha versions mentioned above, no virtualenv). I will stay on `python-can==3.1.1` for now.

Error log for trying to upgrade to `3.2.1a2`:
```
ERROR: Complete output from command python setup.py egg_info:
    ERROR: Download error on https://pypi.org/simple/pytest-runner/: [WinError 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond -- Some packages may not be found!
    Couldn't find index page for 'pytest-runner' (maybe misspelled?)
    Download error on https://pypi.org/simple/: [WinError 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond -- Some packages may not be found!
    No local packages or working download links found for pytest-runner
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\u20e00\AppData\Local\Temp\pip-install-csae2hmk\python-can\setup.py", line 109, in <module>
        tests_require=tests_require
      File "c:\python37\lib\site-packages\setuptools\__init__.py", line 144, in setup
        _install_setup_requires(attrs)
      File "c:\python37\lib\site-packages\setuptools\__init__.py", line 139, in _install_setup_requires
        dist.fetch_build_eggs(dist.setup_requires)
      File "c:\python37\lib\site-packages\setuptools\dist.py", line 717, in fetch_build_eggs
        replace_conflicting=True,
      File "c:\python37\lib\site-packages\pkg_resources\__init__.py", line 782, in resolve
        replace_conflicting=replace_conflicting
      File "c:\python37\lib\site-packages\pkg_resources\__init__.py", line 1065, in best_match
        return self.obtain(req, installer)
      File "c:\python37\lib\site-packages\pkg_resources\__init__.py", line 1077, in obtain
        return installer(requirement)
      File "c:\python37\lib\site-packages\setuptools\dist.py", line 784, in fetch_build_egg
        return cmd.easy_install(req)
      File "c:\python37\lib\site-packages\setuptools\command\easy_install.py", line 673, in easy_install
        raise DistutilsError(msg)
    distutils.errors.DistutilsError: Could not find suitable distribution for Requirement.parse('pytest-runner')
    ----------------------------------------
ERROR: Command "python setup.py egg_info" failed with error code 1 in C:\Users\u20e00\AppData\Local\Temp\pip-install-csae2hmk\python-can\
```

ph4049, 2019-06-13:
I tried out release 3.2.0, 3.2.1a0 and 3.2.1a0.

Since 3.2.0 I run into several issues. First of all 'is_extended_id' causes an error and seems to be unknown. I figured out it's working when replaced with 'extended_id'.

'Send_periodic' isn't working anymore. 
`Attribute Error: 'SocketcanNative_Bus' object has no attribute 'send_periodic'`

Additionally there is an issue with 'modify_data'.
`Attribute Error: 'NoneType' object has no attribute 'modify_data'`

I'm not sure if I might have made a mistake, but my code worked and I've only had these problems since I set up a new raspberry pi with current releases.

Using a rapsberry pi zero w with the waveshare rs485 can hat.

felixdivo, 2019-06-23:
Is this now fixed?

philippherget, 2019-06-24:
Tried it again with all the releases (alpha included) today. Still not working for me.

hardbyte, 2019-06-24:
Sorry I shouldn't have assigned myself - new baby is reducing my open source contributions! :baby: 

AFAIK the original installation issue has been fixed. We need to bump the version and tag it to remove the alpha label.

@philippherget you're able to install though correct? You could be running into problems with the changes to `send_periodic` from the 3.0 release. Could you please open a new issue with details on your operating system, python-can versions that worked and any other relevant info.

phil-marble, 2019-06-24:
@hardbyte , as i noted in https://github.com/hardbyte/python-can/issues/593#issuecomment-494894823, your fix worked for the original issue I reported...

far3ll, 2019-06-26:
With trying to upgrade to python-can v3.2.1 I run into the same error as mentioned above by me. Looks like it cannot find/download a specific file
`ERROR: Download error on https://pypi.org/simple/pytest-runner/: [WinError 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond -- Some packages may not be found!`
Any suggestions?

hardbyte, 2019-06-27:
@far3ll Could you please open a new issue with your OS, Python version and the command that fails?

far3ll, 2019-07-10:
> @far3ll Could you please open a new issue with your OS, Python version and the command that fails?

I don't know if this is even a bug. With installing `pytest-runner` prior to updating `python-can` via pip, everything works fine. I guess it is a combination of a missing dependency (`pytest-runner`) and an internet connection through a proxy.
Cheers!

koberbe, 2019-09-25:
I'm getting the same error for version 3.3.2, when running `pip install python-can==3.3.2`

```
Collecting windows-curses (from python-can==3.3.2)
  ERROR: Could not find a version that satisfies the requirement windows-curses (from python-can==3.3.2) (from versions: none)
ERROR: No matching distribution found for windows-curses (from python-can==3.3.2)
```

pip version: 19.0.3
python version: 3.7.4

airfield20, 2019-10-25:
I had the same issue with version 3.3.2, I used an older version (3.0.0) to complete the install

jcatsong, 2020-03-02:
Any workaround for this?  Trying to install CANOpen v1.0.0 required by another project which hits this issue.

jcatsong, 2020-05-15:
For anyone still hitting this (on Linux at least), I was able to successfully install 3.0.0, but nothing newer.

pip install --user python-can==3.0.0

karlding, 2020-05-16:
@jcatsong can you install the alpha in #715?

```
pip install python-can==3.3.3a3
```

SethDrew, 2020-05-18:
I just tried this and was unable to install 

```pip install python-can==3.3.3a3```

hardbyte, 2020-05-18:
@SethDrew any more information would be appreciated, what went wrong?

karlding, 2020-05-20:
I threw together what I think is a reproducer using the `amd64/ubuntu` Docker image, since I didn't want to install QEMU and fiddle with `binfmt_misc` (but I suspect this will also happen on `arm64v8`).

Seems like the issue is with the default `setuptools` included on 16.04 and support for [Environment Markers](https://www.python.org/dev/peps/pep-0508/).

The following `Dockerfile` fails on 16.04, but passes with the 18.04 image:

```Dockerfile
FROM amd64/ubuntu:16.04

# Update packages and install dependency packages for services
RUN apt-get update \
 && apt-get dist-upgrade -y \
 && apt-get clean \
 && echo 'Finished installing dependencies'

# Install pip
RUN apt-get update && DEBIAN_FRONTEND=noninteractive apt-get install -y \
 python-pip \
&& rm -rf /var/lib/apt/lists/*

# Install python-can
RUN pip install python-can==3.3.3a3
```

But upgrading `setuptools` out of band (on 16.04) works:

```diff
diff --git 1/Dockerfile 2/Dockerfile
--- 1/Dockerfile
+++ 2/Dockerfile
@@ -13,5 +13,8 @@ RUN apt-get update && DEBIAN_FRONTEND=noninteractive apt-get install -y \
  python-pip \
 && rm -rf /var/lib/apt/lists/*

+# Update pip
+RUN pip install --upgrade setuptools
+
 # Install python-can
 RUN pip install python-can==3.3.3a3
```

This makes sense to me, since Ubuntu 16.04 ships with [`setuptools` version `20.7.0`](https://launchpad.net/ubuntu/xenial/+package/python-setuptools) (which was released in [2016](https://github.com/pypa/setuptools/releases/tag/v20.7.0)), however [PEP 508 Environment Marker](https://www.python.org/dev/peps/pep-0508/) support is only in setuptools `36.2.x` (I believe it was implemented in `36.2.0`, broken in `36.2.1`, and then fixed `36.2.2`+), and it seems like people reporting this are running 16.04 (presumably with `pip` and `setuptools` from system packages). Bionic ships with at least [`36.2.7`](https://launchpad.net/ubuntu/bionic/+package/python-setuptools) which is why it works there.

In addition, people are saying that they can only install `3.0.0`, and if we [diff `3.0.0` against `v3.1.0`](https://github.com/hardbyte/python-can/compare/3.0.0..v3.1.0#diff-2eeaed663bd0d25b7e608891384b7298R104) we see the introduction of an Environment Marker, which seems to further support this theory. This seems to be an intentional breakage. 

Updating `setuptools` should allow for newer versions of `python-can` to be installed, but I suppose we can wait for someone to confirm.

```bash
pip install --upgrade setuptools
```

jcatsong, 2020-05-20:
Update after the previous few comments.  I attempted to install 3.3.3a3 per above, failed with the following output.. I'm using Ubuntu 16.04.6 LTS.

> pip install --user python-can==3.3.3a3
Collecting python-can==3.3.3a3
  Downloading https://files.pythonhosted.org/packages/37/13/9daf64b2a1460b5ccb3deecc0cfdd03a73e3d3aa906ee0f00b2ac20ad5ad/python-can-3.3.3a3.tar.gz (176kB)
    100% |████████████████████████████████| 184kB 808kB/s 
Collecting wrapt~=1.10 (from python-can==3.3.3a3)
Collecting aenum (from python-can==3.3.3a3)
  Using cached https://files.pythonhosted.org/packages/6a/c8/1b8513155e4bc68f322e2fbaf90708dd66290a0edaa604ea1675357d18ee/aenum-2.2.3-py2-none-any.whl
Collecting typing; python_version < "3.5" (from python-can==3.3.3a3)
  Using cached https://files.pythonhosted.org/packages/22/30/64ca29543375759dc589ade14a6cd36382abf2bec17d67de8481bc9814d7/typing-3.7.4.1-py2-none-any.whl
Collecting windows-curses; platform_system == "Windows" (from python-can==3.3.3a3)
  Could not find a version that satisfies the requirement windows-curses; platform_system == "Windows" (from python-can==3.3.3a3) (from versions: )
No matching distribution found for windows-curses; platform_system == "Windows" (from python-can==3.3.3a3)
You are using pip version 8.1.1, however version 20.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.

However after updating setup tools per above I was able to upgrade python-can to 3.3.3a3

> pip install --user --upgrade setuptools
Collecting setuptools
  Downloading https://files.pythonhosted.org/packages/ac/d6/0f6c0d9d0b07bbb2085e94a71aded1e137c7c9002ac54924bc1c0adf748a/setuptools-46.4.0.zip (865kB)
    100% |████████████████████████████████| 870kB 1.4MB/s 
Building wheels for collected packages: setuptools
  Running setup.py bdist_wheel for setuptools ... done
  Stored in directory: /home/sean/.cache/pip/wheels/43/1b/2c/595740713473e2078bbda7af8833505d1d5012ec3a1d930371
Successfully built setuptools
Installing collected packages: setuptools
Successfully installed setuptools-46.4.0

> pip install --user python-can==3.3.3a3
Collecting python-can==3.3.3a3
  Using cached https://files.pythonhosted.org/packages/37/13/9daf64b2a1460b5ccb3deecc0cfdd03a73e3d3aa906ee0f00b2ac20ad5ad/python-can-3.3.3a3.tar.gz
Collecting wrapt~=1.10 (from python-can==3.3.3a3)
Collecting aenum (from python-can==3.3.3a3)
  Using cached https://files.pythonhosted.org/packages/6a/c8/1b8513155e4bc68f322e2fbaf90708dd66290a0edaa604ea1675357d18ee/aenum-2.2.3-py2-none-any.whl
Collecting typing (from python-can==3.3.3a3)
  Using cached https://files.pythonhosted.org/packages/22/30/64ca29543375759dc589ade14a6cd36382abf2bec17d67de8481bc9814d7/typing-3.7.4.1-py2-none-any.whl
Building wheels for collected packages: python-can
  Running setup.py bdist_wheel for python-can ... done
  Stored in directory: /home/sean/.cache/pip/wheels/10/6d/7d/a9e8f8cd769f4bbf424b2197b6b05bc01bd64b9f0bdc3ed001
Successfully built python-can
Installing collected packages: wrapt, aenum, typing, python-can
Successfully installed aenum-2.2.3 python-can-3.0.0 typing-3.7.4.1 wrapt-1.12.0

`>>>import can`
`>>> can.__version__`
`'3.3.3-alpha.3'`

hardbyte, 2020-05-22:
@karlding thanks for digging into that. ⭐ 

I'll close this issue and add a note to release 3.3.3 to use an up to date `setuptools`.

PR #629 (2019-06-27): Release v3.3.0
Fixes #593
closed 2020-05-22 by hardbyte