hardbyte/python-can #603: Module object has no attribute 'bus'

Dn1234567, 2019-05-22:
Hello,
python-can updated to the latest version.
Python 2.7
Raspberry (rasbian 9 stretch)

"
import can
bus = can.Bus(interface='socketcan',...
"
gives  AttributeError: 'module' object has no attribute 'bus'
What is wrong?
Thanks in advance

hardbyte, 2019-05-22:
My guess is you could be running your code in a folder which contains a `can.py` which is taking precedence over python-can?

felixdivo, 2019-05-23:
Or a leftover `can.pyc`/`can.pyo`. That was a problem I myself faced once.

Dn1234567, 2019-05-24:
Thanks.
renamed file from can.py to can_start.py. 
But new error :-) reffering to the same line: "'module" object is not callable.

Candump on "can0" works well.
sorry :-( for silly questions

felixdivo, 2019-05-25:
Could you post you entire file for this? It's hard to guess without seeing it. ;)

felixdivo, 2019-06-03:
Please comment if the issue is not solved by the above suggestions. ;)

closed 2019-06-03 by felixdivo

natek512, 2020-01-27:
Hello! Running into the same error as well. 
I've create a virtual environment with the module installed. I tried creating a Bus object via the following

```
import can

bus = can.interface.Bus(bustype='pcan', channel='PCAN_USBBUS1', bitrate=250000)
```

and this is the error I receive:

```
runfile('C:/Documents/Python Scripts/can_api_tutorial.py', wdir='C:Documents/Python Scripts')
Traceback (most recent call last):

  File "C:\Documents\Python Scripts\can_api_tutorial.py", line 17, in <module>
    bus = can.interface.Bus(bustype='pcan', channel='PCAN_USBBUS1', bitrate=250000)

  File "C:\.conda\envs\CANenv\lib\site-packages\can\interface.py", line 127, in __new__
    return cls(channel, *args, **config)

  File "C:\.conda\envs\CANenv\lib\site-packages\can\interfaces\pcan\pcan.py", line 97, in __init__
    self.m_objPCANBasic = PCANBasic()

  File "C:\.conda\envs\CANenv\lib\site-packages\can\interfaces\pcan\basic.py", line 359, in __init__
    self.__m_dllBasic = windll.LoadLibrary("PCANBasic")

  File "C:\.conda\envs\CANenv\lib\ctypes\__init__.py", line 426, in LoadLibrary
    return self._dlltype(name)

  File "C:\.conda\envs\CANenv\lib\ctypes\__init__.py", line 348, in __init__
    self._handle = _dlopen(self._name, mode)

OSError: [WinError 126] The specified module could not be found
```
Pretty new to this so sorry if this is a silly question! Thanks!