hardbyte/python-can #608: Can not log sent messages on KvaserBus

HellDeveloper91, 2019-05-30:
I've seen some issues relating this topic but there is a difference in my case. I'm creating a KvaserBus and a ASCWriter as arguments for a Notifier and I was expecting getting in a ASC file all the can frames running trough the Kvaser but there are only the frames received.
This is a simple demo to reproduce that scenario:

```
import can
import time

from can.interfaces.kvaser import canlib
from can.io import asc

log_listener = asc.ASCWriter(file="dummylog2.asc", channel=0)
virtual_ecu = canlib.KvaserBus(
    channel=0,
    config={'bitrate': 500000, 'receive_own_messages': True}
)
notifier = can.Notifier([virtual_ecu], [log_listener])
supplier_session = can.Message(
    is_extended_id=False,
    arbitration_id=0x7d4,
    data=[0x02, 0x10, 0x60, 0x00, 0x00, 0x00, 0x00, 0x00],
    dlc=8,
    channel=0
)
tester_present = can.Message(
    is_extended_id=False,
    arbitration_id=0x7d4,
    data=[0x02, 0x3e, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00],
    dlc=8,
    channel=0
)
virtual_ecu.send(supplier_session)
time.sleep(1)
virtual_ecu.send_periodic(tester_present, 1)
time.sleep(5)
virtual_ecu.stop_all_periodic_tasks()
virtual_ecu.shutdown()
```
And this is part of the log generated:

>  0.376320 1  516             Rx   d 1 01
>  0.378620 1  7DC             Rx   d 8 02 50 60 00 00 00 00 00
>  1.012990 1  7DC             Rx   d 8 03 7F 3E 13 00 00 00 00
>  1.372840 1  516             Rx   d 1 01
>  2.007890 1  7DC             Rx   d 8 03 7F 3E 13 00 00 00 00
>  2.372780 1  516             Rx   d 1 01
>  3.012860 1  7DC             Rx   d 8 03 7F 3E 13 00 00 00 00
>  3.372710 1  516             Rx   d 1 01
>  4.012880 1  7DC             Rx   d 8 03 7F 3E 13 00 00 00 00
>  4.372650 1  516             Rx   d 1 01
>  5.012820 1  7DC             Rx   d 8 03 7F 3E 13 00 00 00 00
>  5.372670 1  516             Rx   d 1 01

The frames with arbitration_id 7dc are the responses of another device to my sent frames with arbitration_id 7d4 so the frames are being sent but not logged.

Checking the KvaserBus config parameters we can see:

:param bool single_handle:
            Use one Kvaser CANLIB bus handle for both reading and writing.
            This can be set if reading and/or writing is done from one thread.

:param bool receive_own_messages:
            If messages transmitted should also be received back.
            Only works if single_handle is also False.
            If you want to receive messages from other applications on the same
            computer, set this to True or set single_handle to True.

I am assuming that if I create a KvaserBus with the parameter receiver_own_messages True and single_handle False or viceversa I should see my sent messages on the log. Please tell me if I'm doing something wrong or it is an issue. 

I'm using Windows10 and Python 2.7.13, ask me for any info you need.

christiansandberg, 2019-05-31:
That's correct. You should just need to change your Bus constructor to use kwargs instead:

```python
virtual_ecu = canlib.KvaserBus(
    channel=0,
    bitrate=500000,
    receive_own_messages=True
)
```

HellDeveloper91, 2019-05-31:
Yes.. I'm really sorry for that newbie mistake...
It works as expected, thank you really much @christiansandberg !

closed 2019-05-31 by HellDeveloper91