hardbyte/python-can #623: Python-can broken on Raspberry Pi 4.19.42-v7+

skpang, 2019-06-12:
Python-can use to work on the Raspberry Pi but with latest version of the kernel something seems to be broken.

uname -a
Linux raspberrypi 4.19.42-v7+ #1219 SMP Tue May 14 21:20:58 BST 2019 armv7l GNU/Linux

When I try to import can, I get this error.

python3
Python 3.5.3 (default, Sep 27 2018, 17:25:39) 
[GCC 6.3.0 20170516] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import can
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/usr/local/lib/python3.5/dist-packages/python_can-3.2.0-py3.5.egg/can/__init__.py", line 24, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/usr/local/lib/python3.5/dist-packages/python_can-3.2.0-py3.5.egg/can/io/__init__.py", line 9, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 664, in _load_unlocked
  File "<frozen importlib._bootstrap>", line 634, in _load_backward_compatible
  File "/usr/local/lib/python3.5/dist-packages/python_can-3.2.0-py3.5.egg/can/io/logger.py", line 15, in <module>
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 954, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 896, in _find_spec
  File "<frozen importlib._bootstrap_external>", line 1147, in find_spec
  File "<frozen importlib._bootstrap_external>", line 1123, in _get_spec
  File "<frozen importlib._bootstrap_external>", line 1104, in _legacy_get_spec
  File "<frozen importlib._bootstrap>", line 444, in spec_from_loader
  File "<frozen importlib._bootstrap_external>", line 541, in spec_from_file_location
  File "/usr/local/lib/python3.5/dist-packages/python_can-3.2.0-py3.5.egg/can/io/sqlite.py", line 155
    f"INSERT INTO {self.table_name} VALUES (?, ?, ?, ?, ?, ?, ?)"
                                                                ^
SyntaxError: invalid syntax

Anyone know how to fix this ?

skpang, 2019-06-12:
In the sqlite.py file line 154

self._insert_template = (
            f"INSERT INTO {self.table_name} VALUES (?, ?, ?, ?, ?, ?, ?)"
        )
Should the f in front of "INSERT INTO be there ?

There is a Syntax Error.

felixdivo, 2019-06-12:
Yeah, it's meant to be there. It's a [format string](https://realpython.com/python-f-strings/) and was introduced in Python 3.6. It is **not** available on 3.5, and python-can does not support this on the develop branch, as the [Readme](https://github.com/hardbyte/python-can/blob/develop/README.rst) tells (the next version will be 4.0). Please use something like `pip install python-can==3.2+` (I'm not sure about the syntax here).

closed 2019-06-12 by felixdivo

christiansandberg, 2019-06-18:
@felixdivo @hardbyte and others. Have you considered that Debian stable only has Python 3.5 easily installable? Not to mention all slow adapting corporate environments out there. It is probably going to be a lot of people left out in the cold once 4.0 gets released. Is it worth dropping 3.5 yet?