hardbyte/python-can #627: Is it possible to support iso15765-2 message over neovi interface?

zhihongliuus, 2019-06-20:
Here is the example from python_ics:

https://github.com/intrepidcs/python_ics/blob/2201cba0344e923aa6bc2472d5e61c3661b63658/examples/iso15765_example.py#L50

# Iso15765 Fuctions #########################################################
def transmit_iso15765_msg(device, netid=ics.NETID_HSCAN, is_canfd=False):
    number_of_bytes = 64
    msg = ics.CmISO157652TxMessage()
    msg.id = 0x7E0
    msg.vs_netid = netid
    msg.num_bytes = number_of_bytes
    msg.padding = 0xAA
    # Flow Control
    msg.fc_id = 0x7E8
    msg.fc_id_mask = 0xFFF
    msg.flowControlExtendedAddress = 0xFE
    msg.fs_timeout = 0x10 # ms
    msg.fs_wait = 0x3000 # ms
    msg.blockSize = 0
    msg.stMin = 0
    # CmISO157652TxMessage.flags bitfield union isn't implemented as of 2.12, we need to do it manually.
    msg.flags = 0
    # paddingEnable
    msg.flags |= (1 << 5)
    # CANFD: Enable + BRS
    if is_canfd:
        msg.flags |= (1 << 6) | (1 << 7)
    # tx_dl
    msg.flags |= (8 << 23)
    # Data
    msg.data = [x for x in range(number_of_bytes)]

    # Transmit the message
    print("Transmitting iso15765 message on {}...".format(dev_name(device)))
    ics.iso15765_transmit_message(device, netid, msg, 3000)
    # Wait for the messages to be transmitted, this can be calculated a lot better but works here.
    time.sleep((((number_of_bytes/8)*msg.fs_timeout)/1000.0)+0.5)
    #print_message(msg)
    print("Transmitted iso15765 message on {}.".format(dev_name(device)))

felixdivo, 2019-06-24:
Maybe relevant: https://github.com/pylessard/python-can-isotp

karlding, 2019-10-18:
Closing since ISO-TP support is handled by [python-can-isotp](https://github.com/pylessard/python-can-isotp), so there's nothing actionable here. Driver-level support can be handled there.

Please feel free to comment if you feel otherwise.

closed 2019-10-18 by karlding